chipweinberger/flutter_blue_plus #170: How to bifurcate iBeacons and other BLE devices ?

JaimilPatel, 2022-12-01:
Hello @boskokg 

Expected Behaviour :
Actually I want only nearby iBeacon devices while scanning

Actual Behaviour :
Currently its showing all nearby BLE devices along with iBeacon devices while scanning

So please let me know if you have any idea regarding the same

Thank you in advance.

FXschwartz, 2022-12-01:
> Hello @boskokg
> 
> Expected Behaviour : Actually I want only nearby iBeacon devices while scanning
> 
> Actual Behaviour : Currently its showing all nearby BLE devices along with iBeacon devices while scanning
> 
> So please let me know if you have any idea regarding the same
> 
> Thank you in advance.

When you scan you can provide a list of services OR devices that you are looking for and it will only return those devices

closed 2022-12-01 by boskokg

JaimilPatel, 2022-12-02:
Hello @FXschwartz  
Can you please elaborate more like how can I implement that?
Please share code blocks regarding the same