chipweinberger/flutter_blue_plus #171: Not able to Retrieve the list of Scanned and Connected devices.😢

Rachagundla, 2022-12-05:
The Flutter_Blue_Plus plugin not able to detect any nearby available devices. The method getConnectedDevices() from BluetoothManager not list out the available devices.
The mDevices size: 0 from flutter_ble_plus showing zero.

logs: -
D/BluetoothManager(18124): getConnectedDevices
D/FlutterBluePlugin(18124): mDevices size: 0
W/1.raster(18124): type=1400 audit(0.0:785011): avc: denied { search } for name="battery" dev="sysfs" ino=7009 scontext=u:r:untrusted_app:s0:c154,c256,c512,c768 tcontext=u:object_r:sysfs_batteryinfo:s0 tclass=dir permissive=0

lgaljer20, 2022-12-08:
try enabling GPS

qinrunhao, 2023-01-10:
Hi @Rachagundla 
I also encountered this problem. It was clear that two Bluetooth devices were connected, but the method getConnectedDevices() returned 0. Has the host solved it?

gorychama, 2023-01-24:
enable GPS

closed 2023-03-16 by boskokg