chipweinberger/flutter_blue_plus #172: How to get list of nearBy iBeacons using withDevices & withServices

JaimilPatel, 2022-12-06:
Hello 
What should I pass in List of withDevices or withServices argument to get only beacons devices?

 Future startScan({
    ScanMode scanMode = ScanMode.lowLatency,
    List<Guid> withServices = const [],
    List<Guid> withDevices = const [],
    List<String> macAddresses = const [],
    Duration? timeout,
    bool allowDuplicates = false,
  })
class Guid {
  final List<int> _bytes;
  final int _hashCode;
}

tm-topnet, 2022-12-06:
with services are the services inside the bluetooth device (for example the GUID Guid("xxxxx-xx-xxxxxx-...."), so if you know the sppecific GUID of your becaon you can use that.
instead the device is the tipology, the beacons are commonly bluetooth low energy device, so u have to use BluetoothDeviceType.le.

I hope this answer is useful and correct!

nairboon, 2023-01-10:
Note that withDevices currently doesn't do anything. See #183

closed 2023-03-16 by boskokg