chipweinberger/flutter_blue_plus #176: doesn't display names of bluetooth list IOS IPHONE DEVICE,

insysinfo, 2022-12-28:
doesn't display names of bluetooth list IOS IPHONE DEVICE, but on android device displaying Name with Address so user easily with connect and know about connect with which device...
in IOS, we are getting only some ids of bluetooth list also number of devices list(unnecessary devices)

pls check below SS...
https://user-images.githubusercontent.com/70749329/209533013-4a69c1d2-eee3-46eb-b497-464e57bc0778.png

@boskokg  pls help me

abelherl, 2023-01-26:
I have this issue as well, I cannot get the name of this device https://star-m.jp/eng/products/s_print/mpop/index.html

It is found inside the iOS Bluetooth setting page but it doesn't have any name when scanned using flutter_blue_plus. Is there any way this can be fixed? Because I can't even try to connect it by matching the mac address with my Android build because Apple doesn't allow mac address usage.

kevinconcourse, 2023-03-07:
Facing the same issue

closed 2023-03-16 by boskokg

kevinconcourse, 2023-03-16:
What this issue has been closed? @boskokg

boskokg, 2023-03-16:
Hello @kevinconcourse ,
This is not a paid plugin, you have all the code needed to perform investigation in the xcode and see what is the issue with the device. If you want to be fixed, please add PR with the fix.