chipweinberger/flutter_blue_plus #179: Cross platform

lgLindstrom, 2023-01-02:
Is there any plans to support Windows and Linux?

nairboon, 2023-01-10:
Windows & Linux: #6
Mac: #2 

For these, the plugin should probably be converted to a federated plugin: https://docs.flutter.dev/development/packages-and-plugins/developing-packages#federated-plugins

Which has been partially done before by @gocal for the linux platform: [allbrightio/flutter_blue](https://github.com/allbrightio/flutter_blue/tree/federated_plugin)
What needs to be done, is updating & merging these changes into master. Then we would have a semi-federated plugin as a start. 

See also: https://github.com/flutter/flutter/wiki/Plugins-and-Packages-repository-structure

swift-kim, 2023-06-01:
I'm also looking forward to migrating this plugin to the federated plugin architecture.

I'm trying to write a BT plugin for the [Tizen platform](https://github.com/flutter-tizen/plugins) based on this plugin (as a separate, unendorsed package), but it's almost impossible to do so without a stable platform interface. I'd be happy to see a platform interface package released for this plugin.

chipweinberger, 2023-07-20:
Closing since we already have issues for Windows and Linux open.

closed 2023-07-20 by chipweinberger