chipweinberger/flutter_blue_plus #184: Anyone encounter data corruption in BLE transfer ?

koudle, 2023-01-12:
Android APP listens data from BLE likes below:
```
_rCharacteristic?.value.listen((value) {
 ...
}
```
but sometimes receives error data.
By checking the Android bluetooth HCI log, I find Android device receives corrupted data in  BLE transfer.
Does anyone know how to fix it? Thanks very much!

brain9520, 2023-03-16:
You need to explain exactly what the data error is.
But I can say what I encountered:
I encountered bluetooth devices with protocol 4.0 and protocol 5.0 during development. The MTU of 4.0 is generally around 20, which means that only 20 bytes of data can be sent at a time, and the length of data I need to transmit varies, and sometimes even more Many, at this time, if you want to ensure the integrity of the data, you can determine an agreement with the Bluetooth device development engineer. For example, the first few bytes of a piece of data can represent the length of the data to be transmitted. In this case, get the data and analyze it first. The length of the data this time, and then get it, if the length of the obtained data is different from the parsed data, it means that there is a problem with the data this time, then you can apply to the other party to send it again, of course, the length alone is a bit too imprecise, I currently The method used is to judge the CRC value of each piece of data.
The above is an idea to solve data integrity, and then I encountered the reason of incomplete data (my project needs to send up to several M data) during the development process using this framework. Here are some references for you:

// Send large data, such as software upgrades
  void sendDatas(List<int> listCommand) {
    // Set the length of a single data transmission, the _isFive variable is to judge whether it is a 5.0 Bluetooth device
    int singleCommandLen = _isFive ? 245 : 15;
    // If the length of the command sent is greater than the maximum length of a single line, it will be divided into pieces
    if (listCommand.length > singleCommandLen) {
      int tempCount = listCommand.length ~/ singleCommandLen;
      int maxCount = listCommand.length % singleCommandLen == 0
          ? tempCount
          : tempCount + 1;
      List<int> tempList = [];
      var i = 0;
      late Duration dur;
      if (_isFive) {
        // When it is 5.0, the time can be set uniformly to 62 milliseconds
        dur = const Duration(milliseconds: 62);
      } else {
        if (GetPlatform.isAndroid) {
          // If Android 4.0, set 25 milliseconds
          dur = const Duration(milliseconds: 30);
        } else {
          // If it is iOS 4.0, set it to 14 milliseconds
          dur = const Duration(milliseconds: 14);
        }
      }
      Timer.periodic(dur, (timer) {
        int start = i * singleCommandLen;
        tempList = listCommand.sublist(start,
            i == maxCount - 1 ? listCommand.length : start + singleCommandLen);
        _characteristic!.write(tempList, withoutResponse: true);
        if (i == maxCount - 1) {
          // send end
          timer.cancel();
        }
        i++;
      });
    }
  }

The time in the above code may be different depending on the connected Bluetooth device. The time interval I set in the code is relatively stable in the Bluetooth device I use.

koudle, 2023-03-20:
@brain9520 Thanks very much! I appreciate your help so much! 
Through searching with google, I founded out that the corrupted data could be a ble hardware problem. It's very hard to solve with software. So I transmit data with crc value. But I don't know why ble does't guarantee data integrity. Please let me know if you have relevant information.

closed 2023-03-20 by koudle

brain9520, 2023-03-21:
> @brain9520 Thanks very much! I appreciate your help so much! Through searching with google, I founded out that the corrupted data could be a ble hardware problem. It's very hard to solve with [...]

BLE uses radio waves to transmit information, which is to encode data and modulate it into radio frequency signals for transmission. Generally speaking, the radio wave frequency used by BLE is 2.4G, which is a concept of 2.4GHz used by Wi-Fi, wireless keyboard and mouse, and Zigbee, Thread and other protocols. These devices may interfere with each other.
As far as my personal experience is concerned, I worked with my colleagues to develop a Bluetooth upgrade software project. When the distance between the mobile phone and the Bluetooth device (4.0) is too far (more than three meters), some data will be lost. Another point, I don’t know the scene where your Bluetooth device is used. In my actual development, the Bluetooth device only transmits the message transparently and transmits the data reported from the lower computer. In our case, there will also be problems related to the serial port and the Bluetooth baud rate. It will cause data errors. If you are also in this usage scenario, then maybe you can also start from this aspect.

brain9520, 2023-03-21:
刚发现题主也是中国人，还是用中文来说吧？