chipweinberger/flutter_blue_plus #193: [Question] Does using Bluetooth (LE) really needs access to location on iOS?

broiniac, 2023-01-31:
Is it really needed to define in Info.plist those values for iOS?
```
	<dict>  
	    <key>NSLocationAlwaysAndWhenInUseUsageDescription</key>  
	    <string>Need Location permission</string>  
	    <key>NSLocationAlwaysUsageDescription</key>  
	    <string>Need Location permission</string>  
	    <key>NSLocationWhenInUseUsageDescription</key>  
	    <string>Need Location permission</string>
```
I've compared behavior of several Bluetooth-based apps available in AppStore and none of them ask for location to use BLE - I can discover, connect devices as well as read and write to characteristics.

closed 2023-03-16 by boskokg

broiniac, 2023-03-16:
@boskokg Why did you close this without any feedback? Was my question stupid/simple/not related? Or is it just "issue cleaning" campaign?

boskokg, 2023-03-18:
Hello @broiniac 

I have done the cleaning of the issues that noone responded for some time...
People mostly ask for something that is already asked and answered detailed.
Regarding your question, please take an a look at the previously similar closed issues (also check the issues at flutter_blue). If you are unable to find any issue where the same this is asked, please let me know.