chipweinberger/flutter_blue_plus #194: Android no results from scan

Prelly95, 2023-02-01:
Hi everyone,

I am having issues using flutter_blue_plus on an android. I can't seem to get any scan results from calling "scan" or "startScan".
I can reliably read scan results, and connect and communicate on an ios device. however, running the same code on an android device looks to fail silently (i.e. the scan timeout of 10 seconds elapses without finding a single ble device).

I have enabled all the required permissions in the AndroidManifest.xml file, and the permissions seem to have taken effect as the app asks for location permission when "scan" is called. Also going into the app settings and viewing all permission reveals that Bluetooth scanning and connecting is allowed.

The following is a minimal implementation that reproduces the problem:

```import 'package:flutter/material.dart';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';

class MyApp extends StatelessWidget {
  const MyApp({super.key});

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      home: Center(
        child: ElevatedButton(
            onPressed: () async {
              FlutterBluePlus flutterBlue = FlutterBluePlus.instance;
              flutterBlue.scan(timeout: Duration(seconds: 10)).listen(
                (r) {
                  print('${r.device.name} found! rssi: ${r.rssi}');
                },
              );
              print("Done");
            },
            child: Text("Scan")),
      ),
    );
  }
}
```

I really hope this isn't a "works on my machine" problem.
Do you have any recommendations to solve the problem or is there anyone who faced the same problem?

Cheers!
Pat

Prelly95, 2023-02-01:
Oh, and this is the output from the terminal when the button is pressed.

```
D/ViewRootImpl@f550c30[MainActivity](26850): ViewPostIme pointer 0
D/ViewRootImpl@f550c30[MainActivity](26850): ViewPostIme pointer 1
I/flutter (26850): Done
D/BluetoothAdapter(26850): STATE_ON
I/chatty  (26850): uid=10212(com.xxxxxx.xxxxx) identical 1 line
D/BluetoothAdapter(26850): STATE_ON
D/BluetoothLeScanner(26850): Start Scan with callback
D/BluetoothLeScanner(26850): onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2
D/BluetoothAdapter(26850): STATE_ON
D/BluetoothLeScanner(26850): Stop Scan with callback
```

simon-iversen, 2023-02-01:
I have the same issue on my Android phone. Not sure if it's related but it seems like this started to fail after I ran a firmware update https://www.samsung.com/uk/support/mobile-devices/what-are-the-new-features-and-upgrades-of-one-ui-4-1/

I have not found a solution yet.

simon-iversen, 2023-02-01:
I resolved this now. I had to do the following:

- Go to settings
- Location
- App permissions
- Find your app and click on it
- Check "Allow only while using the app" and "Use precise location"

Prelly95, 2023-02-02:
Thanks heaps for your help on this, it seems to have solved the problem 👍

closed 2023-02-02 by Prelly95