chipweinberger/flutter_blue_plus #197: Remove from iOS only

RogerBrusamarello, 2023-02-03:
Apple is refusing my App because I remove the information in Info.plist
They returned this: ITMS-90683: Missing purpose string in Info.plist.

There is an option to remove implementation on my App in iOS, and keeping in Android?

broiniac, 2023-02-04:
@RogerBrusamarello Can you rephrase that? What do you want to achieve at the end?
Also - which purpose string is missing in Info.plist?

RogerBrusamarello, 2023-02-04:
I want run this package only in Android not in iOS.

My bluetooth devices is no ready for iOS, so I would like delivery in Apple Store. But Always I get this error on Info.plist because i did not put the purpose strings in it.

ViliusP, 2023-02-09:
You could try to use environment variables and conditional rendering:

```dart
static const bool ANDROID_PLATFORM = bool.fromEnvironment("ANDROID_PLATFORM", defaultValue: false);
...
  @override
  Widget build(BuildContext context) {
     // Android-specific code with Bluetooth functionality
     if (ANDROID_PLATFORM) return androidWidget;
     // Should contain no flutter_blue_plus references
     return defaultWidget;
  }
...
```

When building apk/appbundle use `...--dart-define=ANDROID_PLATFORM=TRUE` and it **SHOULD** be tree shaken. Look at https://github.com/flutter/flutter/issues/23122.

closed 2023-03-16 by boskokg