chipweinberger/flutter_blue_plus #198: Foreground Discover Services: no bluetooth gatt

ZainNomancowlar, 2023-02-06:
Im using the [flutter foreground task plugin](https://pub.dev/packages/flutter_foreground_task) and im getting some weird behaviour. The foreground service automatically connects to a particular device. In the Flutter/UI code I periodicaly call `FlutterBluePlus.instance.connectedDevices` to get the connected devices and the device shows up as a connected device. But when i try to get the sevices of the bluetooth device i get the error 
`discover_services_error, no instance of BluetoothGatt, have you connected first?`.  There is other wierd behaviour at times as well but I havent narrowed down when exactly that occurs.

lorenzosfarra, 2023-06-28:
Hi,

hope I am not off-topic here, but I have started from the same problem.
If it's off-topic, I will open a different issue for discussions/suggestions, but I think your opinion will not only assist me, but also benefit other developers who are seeking a reliable and solid strategy for managing foreground services and background logic related to BLE.

When it comes to foreground services and UI code, my main idea was to:

  - UI:  
    - getting a deviceId with `FlutterForegroundTask.getData<String>(key: 'deviceId')`. If present, basically do nothing
    - not present ? open a dialog to perform the scan of the devices, show them and save the deviceId when the user selects it with `FlutterForegroundTask.saveData(key: "deviceId", value: deviceId)`.
    
 - start the foreground service
     - getting a deviceId with `FlutterForegroundTask.getData<String>(key: 'deviceId')`. 
       - present? Connect to it and work with services and characteristics I already know (fixed IDs)
       - not present ? wait until the next onRepeatEvent to do the same

Thanks and have a nice day,
  Lorenzo

chipweinberger, 2023-07-20:
please try 1.7.6+, feel free to reopen if you still have problems.

closed 2023-07-20 by chipweinberger