chipweinberger/flutter_blue_plus #202: Support Apearance values

MrCsabaToth, 2023-02-08:
There are Categories and Subcategories defined here 
[Appearance Values.pdf](https://github.com/boskokg/flutter_blue_plus/files/10681601/Appearance.Values.pdf)

In some cases it's not enough to see what services a device exposes to exactly identify them. For example FTMS Fitness Machines hint their types (Indoor Bike, Treadmill, Rower, ...) via advertisement service data.

However in case of CSC (Cycling Speed and Cadence Sensors) there's no such service data, but I could tell apart speed and cadence sensors by looking at the appearance. That would hint the type. I saw that nRF Connect can sense that appearance without connecting to a device and polling the characteristics.

Bellow is an nRF scan, this time with a HRM:
[image]

MrCsabaToth, 2023-02-10:
Looks like for example the Service Data coming over the advertisement has a data type 0x16, whereas the appearance has a data type 0x19. The appearance is a fixed two byte value. This gonna be https://developer.android.com/reference/android/bluetooth/le/ScanRecord#DATA_TYPE_APPEARANCE
As far as I can see there's no direct call on the Android API unfortunately, so we'd need to call https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getAdvertisingDataMap() and look into that if it has 0x19 datatype and extract the appearance if it has any. As for iOS I don't know, currently I don't need that, someone else would need to help with that. But I think this is lacking from the Flutter Blue interface, the appearance is the common way to define "icons" representing the various devices during scan.

MrCsabaToth, 2023-02-10:
Oh, I found the place, the `static AdvertisementData parse(byte[] rawData) {` in the `AdvertisemetnParser.java` has a swtich based on the advertisement data type.

MrCsabaToth, 2023-02-10:
I tested my PR but the appearance is not picked up (stays null). But this shines a light to another problem of mine: all of the Tx Power Level data is null as well. This is with an Android 12 device.

MrCsabaToth, 2023-02-10:
I tried to make both the txPowerLevel and appearance be non nullable (both defaulting to 0), however I still cannot read any tx power level or appearance.

MrCsabaToth, 2023-07-18:
I'll completely rework the PR now that ProtoBuf is gone.

chipweinberger, 2023-09-30:
iOS does not support appearance values in scan advertisements.

existing code:
```
- (NSDictionary *)bmScanResult:(CBPeripheral *)peripheral
             advertisementData:(NSDictionary<NSString *, id> *)advertisementData
                          RSSI:(NSNumber *)RSSI
{
    NSString     *localName      = advertisementData[CBAdvertisementDataLocalNameKey];
    NSNumber     *connectable    = advertisementData[CBAdvertisementDataIsConnectable];
    NSNumber     *txPower        = advertisementData[CBAdvertisementDataTxPowerLevelKey];
    NSData       *manufData      = advertisementData[CBAdvertisementDataManufacturerDataKey];
    NSArray      *serviceUuids   = advertisementData[CBAdvertisementDataServiceUUIDsKey];
    NSDictionary *serviceData    = advertisementData[CBAdvertisementDataServiceDataKey];
```

because of this, I think we should close the issue.

I always avoid adding platform specific features if the other platform cannot support it.

closed 2023-09-30 by chipweinberger

MrCsabaToth, 2023-10-02:
It's kinda unfortunate that iOS doesn't support this. However the feature is extremely important, since you cannot tell apart certain devices without actually connecting to them (which is infeasible).
I normally would agree to try to support features which both platforms support, however this is one case where I'd have to fork the repository and implement it for myself.

chipweinberger, 2023-10-02:
if you really want it, open a pr and i'll consider it for sure

MrCsabaToth, 2024-01-22:
It's sad that iOS is so restrictive. FBP has several features which iOS doesn't support, like some MTU related stuff. I'll definitely craft a PR.

PR #772 (2024-01-23): [Feature] Adding Appearance support (part of advertisement data)
Fixes #766 and Fixes #202