chipweinberger/flutter_blue_plus #205: Is Cancelling in-progress pairing implemented?

LeoBound, 2023-02-10:
Is it possible to cancel a connection to the device mid-pair?
Sometimes the pairing process takes a long time so I would like to allow the user to cancel it and try again/select another device.

It appears on the Android side of things there is BluetoothConnectionManager.cancelPairing() and for iOS there is CBCentralManager.cancelPeripheralConnection() so if it's not already possible with flutter_blue_plus it may be implementable?

FXschwartz, 2023-02-10:
> Is it possible to cancel a connection to the device mid-pair? Sometimes the pairing process takes a long time so I would like to allow the user to cancel it and try again/select another device.
>
> It [...]

Why can't you just set the timeout on the pair method? That way if it takes longer than the Duration you pass it will cancel the pairing.

LeoBound, 2023-02-10:
Aa far as I know this doesn't cancel the Future's computation, it just stops waiting for it (and throws a TimeoutException).
So after timing out Android continues the connection process and can eventually connect to the device that was started when calling the async function.
I think this is why dart has CancelableOperation/CancelableCompleter.
I may be wrong on this though, but I'm fairly sure I've seen this behaviour before (after timing out the device.connect() call).

closed 2023-03-16 by boskokg