chipweinberger/flutter_blue_plus #209: Exception: Failed to write the characteristic on Android 13

sakina1403, 2023-02-16:
Hello, thanks for the library.

I am using this library in combination with esp_provisioning and trying provision my device via BLE, i.e by sending WiFi credentials over Bluetooth. It works well on Android 10 and I believe up to android 12. But starts failing on Android 13. This may be related to characteristic length but I don't get any error message except this exception: "Failed to write the characteristic".

this might be helpful https://stackoverflow.com/questions/73772158/bluetoothgattcharacteristic-write-crashes-with-android-13

Please check, its completely unusable now on the latest devices and not able to resolve this. If you have any suggestions, I would be happy to modify the library code at my side and try it out.

sakina1403, 2023-02-16:
Most likely related to this: https://github.com/boskokg/flutter_blue_plus/issues/154#issue-1422065134

pktwins, 2023-02-19:
Can you able fetch bluetooth devices list in android 12 and android 13.
For below versions its working fine .

karansingla007, 2023-02-23:
Hello @sakina1403 it's not working in Android 10 also.
I am stuck because of this.

```
E/flutter ( 2750): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter ( 2750): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
E/flutter ( 2750): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #2      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:117:18)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #3      DeviceScreen._buildServiceTiles.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:flutter_blue_plus_example/main.dart:208:29)
```

```
E/flutter ( 2750): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(read_characteristic_error, unknown reason, may occur if readCharacteristic was called before last read finished., null, null)
E/flutter ( 2750): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
E/flutter ( 2750): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #2      BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:80:5)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #3      DeviceScreen._buildServiceTiles.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:flutter_blue_plus_example/main.dart:201:40)
```

pktwins, 2023-02-24:
For android 12 & 13 if we give location permission also it’s working fine. For Upto android 11 it’s worked well. You need to check your Bluetooth device write permission then only you can write.

closed 2023-03-16 by boskokg

3ph, 2023-05-16:
Hi @boskokg, the issue still persist. Why was it closed?