chipweinberger/flutter_blue_plus #211: Experiencing unreliable connectivity

kbessemer, 2023-02-21:
I am trying to connect to a raspberry pi we have turned into a bluetooth device. My app with flutter blue plus is able to connect to another bluetooth device without any issues, and receive all services/characteristics. But when attempting to connect to our raspberry pi it is unreliable, and will only work maybe 5% of the time.

Before you blame our raspberry pi, please understand that we have used other BLE apps in the android store and we're able to connect to our raspberry pi device over bluetooth without any issues.

With my limited bluetooth experience I am not sure what the possible problems or causes for this could be.

Phone: Samsung Galaxy S21 FE
Android: 12

Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.3.0, on macOS 13.0.1 22A400 darwin-arm, locale en-US)
[✓] Android toolchain - develop for Android devices (Android SDK version 33.0.0-rc2)
[✓] Xcode - develop for iOS and macOS (Xcode 14.2)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2021.2)
[✓] VS Code (version 1.63.2)
[✓] Connected device (3 available)
[✓] HTTP Host Availability

• No issues found!

minSdkVersion 19

<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/

void ConnectToDevice(ScanResult device) async {
    setState(() {
      loading = true;
    });
    await device.device.connect();
    var listening = device.device.state.listen((state) async {
      print(state);
      if (state == BluetoothDeviceState.connected) {
        setState(() {
          loading = false;
          connected = true;
        });
        ScaffoldMessenger.of(context).showSnackBar(SnackBar(content: Text(
                     "Connection Established",
                     style: TextStyle(fontWeight: FontWeight.bold, color: Colors.black)),
                     backgroundColor: Colors.grey));
        List<BluetoothService> services = await device.device.discoverServices();
        Navigator.push(
            context,
            MaterialPageRoute(
                builder: (context) =>
                    ConnectionResultsView(services, device)
            )
        );
      }
    });
  }

kevindowling, 2023-02-28:
what model raspi?

kbessemer, 2023-02-28:
> what model raspi?

Raspberry Pi 4 Model B Rev 1.4

anthonyhunter, 2023-03-03:
You might try setting your connection state handler before calling connect, less chance of missing connect, especially since you are blocking on connect.

closed 2023-03-16 by boskokg