chipweinberger/flutter_blue_plus #212: Difference between .scan and .startScan

novvia-dev, 2023-02-28:
I am not sure when you would use one over the other and the examples only show startScan, but the docs say to "use scan" in some cirucumstances.

THoughts?

FXschwartz, 2023-02-28:
> I am not sure when you would use one over the other and the examples only show startScan, but the docs say to "use scan" in some cirucumstances.
> 
> THoughts?

If you look at the source code they are well documented with comments.

```
  scan()
  /// Starts a scan for Bluetooth Low Energy devices and returns a stream
  /// of the [ScanResult] results as they are received.
  ///
  /// timeout calls stopStream after a specified [Duration].
  /// You can also get a list of ongoing results in the [scanResults] stream.
  /// If scanning is already in progress, this will throw an [Exception].
```

``` 
  startScan()
  /// Starts a scan and returns a future that will complete once the scan has finished.
  ///
  /// Once a scan is started, call [stopScan] to stop the scan and complete the returned future.
  ///
  /// timeout automatically stops the scan after a specified [Duration].
  ///
  /// To observe the results while the scan is in progress, listen to the [scanResults] stream,
  /// or call [scan] instead.
```

novvia-dev, 2023-02-28:
Thank you @FXschwartz I also found that there was a random stopScan() in the code, which had me thinking the .scan was not working.

closed 2023-02-28 by novvia-dev