chipweinberger/flutter_blue_plus #214: On Android , turn off Bluetooth, turn it on again, and call the connect device

codepgq, 2023-03-02:
```
D/FlutterBluePlugin( 2180): mDevices size: 1
D/BluetoothAdapter( 2180): isLeEnabled(): ON
D/BluetoothLeScanner( 2180): could not find callback wrapper
D/BluetoothAdapter( 2180): isLeEnabled(): ON
D/BluetoothLeScanner( 2180): onScannerRegistered() - status=0 scannerId=13 mScannerId=0
D/BluetoothAdapter( 2180): isLeEnabled(): ON
E/BluetoothGatt( 2180): android.os.DeadObjectException
E/BluetoothGatt( 2180): 	at android.os.BinderProxy.transactNative(Native Method)
E/BluetoothGatt( 2180): 	at android.os.BinderProxy.transact(BinderProxy.java:571)
E/BluetoothGatt( 2180): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.clientConnect(IBluetoothGatt.java:2546)
E/BluetoothGatt( 2180): 	at android.bluetooth.BluetoothGatt.connect(BluetoothGatt.java:1023)
E/BluetoothGatt( 2180): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$3$com-boskokg-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:344)
E/BluetoothGatt( 2180): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda5.op(Unknown Source:6)
E/BluetoothGatt( 2180): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.ensurePermissionBeforeAction(FlutterBluePlusPlugin.java:756)
E/BluetoothGatt( 2180): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:317)
E/BluetoothGatt( 2180): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
E/BluetoothGatt( 2180): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/BluetoothGatt( 2180): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
E/BluetoothGatt( 2180): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/BluetoothGatt( 2180): 	at android.os.Handler.handleCallback(Handler.java:938)
E/BluetoothGatt( 2180): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/BluetoothGatt( 2180): 	at android.os.Looper.loopOnce(Looper.java:201)
E/BluetoothGatt( 2180): 	at android.os.Looper.loop(Looper.java:288)
E/BluetoothGatt( 2180): 	at android.app.ActivityThread.main(ActivityThread.java:7911)
E/BluetoothGatt( 2180): 	at java.lang.reflect.Method.invoke(Native Method)
E/BluetoothGatt( 2180): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
E/BluetoothGatt( 2180): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1009)
I/flutter ( 2180): catch some error: PlatformException(reconnect_error, error when reconnecting to device, null, null)

```

code:
```
try {
  await disconnectAll();
await stopScan();
startScan();
var  device = await scanResults
.timeout(
            const Duration(seconds: 10),
            onTimeout: (e) => e.addError('search timeout'),
          )
          .map((value) {
            MeDevice? device;
            for (var dev in value) {
               if (macAddressList.contains(dev.macAddress)) {
                 device = dev;
                 break;
               }
            }
            return device;
          })
          .where((event) => event != null)
          .cast<MeDevice>()
          .first;

device.connect();
} catch(e) {
}

```

RodrigoLScorza, 2023-03-08:
@codepgq me too... I dont know why...

E/BluetoothGatt( 8644): android.os.DeadObjectException
E/BluetoothGatt( 8644): 	at android.os.BinderProxy.transactNative(Native Method)
E/BluetoothGatt( 8644): 	at android.os.BinderProxy.transact(Binder.java:763)
E/BluetoothGatt( 8644): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.clientDisconnect(IBluetoothGatt.java:1387)
E/BluetoothGatt( 8644): 	at android.bluetooth.BluetoothGatt.disconnect(BluetoothGatt.java:960)
E/BluetoothGatt( 8644): 	at android.bluetooth.BluetoothGatt$2.onBluetoothStateChange(BluetoothGatt.java:681)
E/BluetoothGatt( 8644): 	at android.bluetooth.IBluetoothStateChangeCallback$Stub.onTransact(IBluetoothStateChangeCallback.java:55)
E/BluetoothGatt( 8644): 	at android.os.Binder.execTransact(Binder.java:682)

codepgq, 2023-03-08:
After I tried to clean up Gatt，is work now for me. @RodrigoLScorza 
```dart
  device.state.skip(1).listen((event) {
      if (event == BluetoothDeviceState.disconnected && !isConnecting) {
        /// clear device's gatt
        device.clearGattCache().then((value) {
          /// remove device from mDevices
          device.disconnect();
        }).catchError((_){});
      }
    });
```

after I think clean gatt, device should be auto remove device form mDevices, or else `connect` will be throw error

zhaojun1217, 2023-03-09:
> After I tried to clean up Gatt，is work now for me. @RodrigoLScorza
>
> ```dart
>   device.state.skip(1).listen((event) {
>       if (event == BluetoothDeviceState.disconnected && !isConnecting) { [...]

good idea !   simple and effective     it help me

thewh1teagle, 2023-03-16:
@codepgq 
Can you explain what you did?
and where !isConnecting comes from?

chipweinberger, 2023-04-28:
merged a fix here: https://github.com/boskokg/flutter_blue_plus/commit/33f0c4c5268c072d357e01facd6bc2a3369eea81

chipweinberger, 2023-04-28:
actually I think my change will not fix it. we need to do it on disconnect.

chipweinberger, 2023-07-20:
please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger