chipweinberger/flutter_blue_plus #221: Handle disconnection

thewh1teagle, 2023-03-15:
When I listening to character of service and bluetooth is turned off or the bluetooth device is turned off so I got disconnection, it doesn't reconnect, and the listen method of the character doesn't throws error
how should I handle that?

thewh1teagle, 2023-03-15:
[image]
this is what happening after the device is disconnected from my phone but I don't know where to catch that

brain9520, 2023-03-16:
device.state.listen((event) {
      // Here you can monitor the Bluetooth connection status
      if (event == BluetoothDeviceState.disconnected) {
        _isDisconnected = true;
      }
      if (event == BluetoothDeviceState.connected) {
        _isDisconnected = false;
      // Here you can reconnect the device
      }
    });

alexvnesta, 2023-03-16:
@brain9520 I had this same question - Thanks so much for your response! Also, good timing. Asked the question just hours before I needed the solution :)

brain9520, 2023-03-16:
> @brain9520 I had this same question - Thanks so much for your response! Also, good timing. Asked the question just hours before I needed the solution :)

You are welcome, it is my honor to help you, I think this is also the meaning of sharing。

closed 2023-03-16 by boskokg