chipweinberger/flutter_blue_plus #225: Example not picking up any devices on Android

sweetfa, 2023-03-20:
Running the example app on Android Samsung SMA725S does not scan any devices.

Bluetooth permissions are requested and approved when running the app, however the logs show `D/FlutterBluePlugin(17374): mDevices size: 0`

Am running in debug mode using Android Studio with flutter version

Flutter 3.8.0-14.0.pre • channel master • https://github.com/flutter/flutter.git
Framework • revision dad9eb21f9 (4 weeks ago) • 2023-02-22 07:16:23 +0800
Engine • revision 3aa112c75c
Tools • Dart 3.0.0 (build 3.0.0-261.0.dev) • DevTools 2.21.1
frank@Bush-Life-2020 example % 

Android VERSION  12

The example app is working on an iOS version.

Does anyone know of any reason why Android is failing to pick up devices that the iOS device can pick up?

anuj9196, 2023-03-21:
Facing same issue

brain9520, 2023-03-21:
Confirm whether the AndroidManifest.xml file contains the following permission statement:
``` xml
    <!-- Allow Bluetooth -->
    <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />
    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />
    <!-- Needed only if your app looks for Bluetooth devices.
        If your app doesn't use Bluetooth scan results to derive physical
        location information, you can strongly assert that your app
        doesn't derive physical location. -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <!-- Needed only if your app makes the device discoverable to Bluetooth
         devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
    <!-- Needed only if your app communicates with already-paired Bluetooth
      devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <!-- If your application supports a service and can run on Android 10 (API level 29) or Android 11, you must also declare 
    the ACCESS_BACKGROUND_LOCATION permission to discover Bluetooth devices. -->
    <!-- <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" /> -->
```
Then you need to dynamically declare the above permissions, please make sure that the dynamic permissions have been applied for and successfully obtained.

sweetfa, 2023-03-21:
I am using the flutter_blue_plus example app!!!

renyun2, 2023-03-21:
AndroidManifest.xml：

<uses-permission android:name="android.permission.BLUETOOTH_SCAN"
  android:usesPermissionFlags="neverForLocation" />

brain9520, 2023-03-21:
> Confirm whether the AndroidManifest.xml file contains the following permission statement:
> 
> ```
>     
>     <uses-feature
>         android:name="android.hardware.bluetooth_le"
>         android:required="true" />
>     
>     <uses-permission android:name="android.permission.BLUETOOTH"
>         android:maxSdkVersion="30" />
>     <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
>         android:maxSdkVersion="30" />
>     
>     <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
>     
>     <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
>     
>     <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
>     
>     
> ```
> 
> Then you need to dynamically declare the above permissions, please make sure that the dynamic permissions have been applied for and successfully obtained.
> 
> For example: I use [permission_handler: ^10.2.0](https://pub.dev/packages/permission_handler) to apply for permission:
> 
> ```dart
>   List<Permission> permissions = [
>         Permission.location,
>         Permission.storage,
>         Permission.bluetooth,
>         Permission.bluetoothConnect,
>         Permission.bluetoothScan,
>         ];
>    await permissions.request();
> ```

[image]

sweetfa, 2023-03-21:
[image]

I am on master branch.  Should I be on another branch as the manifest file does not contain any permissions on master branch

sweetfa, 2023-03-21:
[image]

sweetfa, 2023-03-21:
Ok, if I add the permissions to the AndroidManifest.xml in android/app/src/main

and

increase the compileSdkVersion to 33 in android/app/build.gradle

and

add the following to Main.dart

`void main() {
  WidgetsFlutterBinding.ensureInitialized();
  <Permission>[
    Permission.location,
    Permission.storage,
    Permission.bluetooth,
    Permission.bluetoothConnect,
    Permission.bluetoothScan
  ].request().then((status) {
    runApp(const FlutterBlueApp());
  });
}

and add permission_handler: ^10.2.0 to pubspec.yaml

it works.

I don't appear to be out of sync with the main branch, so perhaps some changes are still to be merged to that branch.

sweetfa, 2023-03-21:
Thanks for your help.  You got me past my hurdle.  I presume someone will create a task to update the master branch so that it has the correct permissions.

brain9520, 2023-03-21:
Is your problem solved? just took a nap.

sweetfa, 2023-03-21:
Yes, my problem is solved, however the example app as I have received it from github still has the issue.

chipweinberger, 2023-04-27:
fixed: https://github.com/boskokg/flutter_blue_plus/commit/5ed330a73efbf64edda47bd92c29e4bca74f9f62

let me know if it looks right.

closed 2023-04-27 by chipweinberger

sweetfa, 2023-04-28:
Thanks Chip
The example now works on Android at the commit you specified.

In case you are not aware, the head of master currently does not compile.

Yolo-cell-hash, 2024-10-02:
I'm facing a similar issue despite of following all the above mentioned steps ... is there any change to be made in the permissions as of now ?

My current state is - 

  <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />

    <uses-permission android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />

    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />

    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />

    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />

Do I need to add anything other than this ? My use case is simply to print available devices but as of now I cannot detect anything other than watches, airpods,etc on my Android

brain9520, 2024-10-10:
> I'm facing a similar issue despite of following all the above mentioned steps ... is there any change to be made in the permissions as of now ?
> 
> My current state is -
> 
> ```
> <uses-permission android:name="android.permission.BLUETOOTH"
>     android:maxSdkVersion="30" />
> 
> <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
>     android:maxSdkVersion="30" />
> 
> <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
> 
> <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
> 
> <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
> 
> <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
> 
> <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
> ```
> 
> Do I need to add anything other than this ? My use case is simply to print available devices but as of now I cannot detect anything other than watches, airpods,etc on my Android

There is a parameter in the scanning method that is only for the Android platform: androidScanMode. By default, it does not need to be set and the default setting is low power consumption. According to your description, you can scan BLE devices. I guess the parameters here need to be set according to the situation.

chipweinberger, 2024-10-10:
>  I cannot detect anything other than watches, airpods,etc on my Android

If you can scan *anything*, then you have already have scanning permissions. It's not a permissions issue.