chipweinberger/flutter_blue_plus #226: read_characteristic_error, unknown reason, may occur if readCharacteristic was called before last read finished

kbessemer, 2023-03-23:
When I connect to my peripheral the first time, the characteristic is read successfully and my UI gets updated correctly. This is for battery levels. But if I disconnect from the peripheral, then reconnect, and attempt to read from the same characteristic, I get this error:

I/flutter (12535): PlatformException(read_characteristic_error, unknown reason, may occur if readCharacteristic was called before last read finished., null, null)

Here is another error and some debugging messages going into the console:

I/flutter (12535): LateInitializationError: Field 'battery_bytes' has not been initialized.
I/flutter (12535): BATTERY [54]
I/flutter (12535): Setting battery state
I/flutter (12535): 54

You can see I am getting the battery level from the peripheral. I think the error is that this is asynchronous, and I am not sure how to change my code to deal with it.

`
  void BuildServiceList() async {
    List service_widgets = [];

    for (var service in widget.services) {
      if (service.uuid.toString().toLowerCase().contains("18ae")) {
        print("Found SERVICE with 18AE");
        for (var char in service.characteristics) {
          print(char);
          if (char.uuid.toString().toLowerCase().contains("fa01")) {
            print(char);
            print("Found CHAR with FA01");
            if (char.properties.read) {
              c_read_uart = char;
            }
          }
          if (char.uuid.toString().toLowerCase().contains("fa03")) {
            print(char);
            print("Found CHAR with FA03");
            if (char.properties.write) {
              c_write_uart = char;
            }
          }
        }
      }
      if (service.uuid.toString().toLowerCase().contains("180f")) {
        print("Found SERVICE with 180F");
        for (var char in service.characteristics) {
          print(char);
          if (char.uuid.toString().toLowerCase().contains("2a19")) {
            print("Found CHAR with 2A19");
            if (char.properties.read) {
              c_read_battery = char;
            }
          }
        }
      }
      List char_widgets = [];
    }
    Timer(Duration(seconds: 4), ToggleLoading);
    batteryTimer = await Timer(Duration(seconds: 4), ReadBattery);
  }

  void ReadBattery() async {
    try {
      battery_bytes = await c_read_battery.read();
    } catch(e) {
      print(e);
    }
    await Timer(Duration(seconds: 2), SetBatteryLevel);
  }

  void SetBatteryLevel() {
    try {
      print("BATTERY " + battery_bytes.toString());
      setState(() {
        print("Setting battery state");
        batteryValue = battery_bytes[0];
      });
      print(batteryValue);
    } catch(e) {
      print(e);
    }
  }
`

kbessemer, 2023-03-28:
I found the cause of the issue. The connection listener will send you to the next screen (the device/peripheral landing screen) when it gets a successful connection message, for some reason when connecting to the device for the 2nd or 3rd time we were getting duplicate success messages on the connection listener, which was creating duplicate device landing screens being built on top of eachother..all of which were attempting to read from this battery characteristic at the same time. I resolved this in the connection listener code which sends the user to the next screen.

```
var listening = device.device.state.listen((state) async {
      print("listening to device");
      print(state);
      if (!connected) {
        if (state == BluetoothDeviceState.connected) {
          setState(() {
            loading = false;
            connected = true;
          });
          Provider
              .of<AppData>(context, listen: false)
              .changeConnected(true);
          if (Platform.isAndroid) {
            await device.device.requestMtu(156);
          }
          List<BluetoothService> services = await device.device
              .discoverServices();
          Provider
              .of<AppData>(context, listen: false)
              .changeDeviceName(device_name);
          pushNewScreen(
            context,
            screen: DeviceLandingView(services, device, device_name),
            withNavBar: false, // OPTIONAL VALUE. True by default.
            pageTransitionAnimation: PageTransitionAnimation.cupertino,
          );
        }
      }
    });
```

chipweinberger, 2023-04-28:
great. glad to hear you fixed it.

closed 2023-04-28 by chipweinberger