chipweinberger/flutter_blue_plus #227: HC-05 not visible even after scanning.

MohsinP07, 2023-03-24:
In my project I want to connect my HC-05 to the device but the device is not visible even after scanning multiple times. I am able to view other devices but not HC-05. Is there any solution for this?

xbug42, 2023-04-21:
HC-05 is not a BLE module so it won't work. 
You have to use another bluetooth module (HC-08, AT-09, ...) or another plugin (BluetoothSerial but it's only for Android... afaik, IOS only support BLE devices)

MohsinP07, 2023-04-21:
Hey @xbug42 thanks for commenting,
Currently I am using hm10 module and I can now see the device after scan and connect it. 
But the data which I am receiving from the device is of 30 characters string, and I am not receiving the data properly means not in sequence. On every instance I call the function for receiving the data, different data is received. 
The device from which I am receiving the data sends the data sends data continuesly.

chipweinberger, 2023-07-20:
Flutter Blue Plus is BLE only.

reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger