chipweinberger/flutter_blue_plus #231: Update Read.me for potential problem in android release mode

Clement-Marchais-Wiztivi, 2023-04-05:
A lot of person have some difficulty to run the plugin in release mode on Android 
As mentionned is this [StackOverflow comment](https://stackoverflow.com/questions/71291815/flutter-blue-plus-is-not-working-on-release-mode#:~:text=Tell%20ProGuard%20to%20ignore%20the%20flutter_blue(_plus)%20files.), you may need to add this comment in the section **Troubleshooting**': 

Tell ProGuard to ignore the flutter_blue_plus files. Add a proguard-rules.pro file to your apps folder
Make sure the file is referenced in your apps build.gradle e.g. as this

```
android {
  buildTypes {
    release {
      minifyEnabled true
      shrinkResources true
      proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
    }
  }
}
```

Actually tell ProGuard what to ignore inside the pro guard-rules.pro file. Pick the line with the library you are actually using.
```
# for flutter_blue_plus
-keep class com.boskokg.flutter_blue_plus.** { *; }
```

What do you think ?

chipweinberger, 2023-04-27:
please open a PR. thanks.

chipweinberger, 2023-07-20:
protobuf has now been removed in 1.7.0.

closed 2023-07-20 by chipweinberger