chipweinberger/flutter_blue_plus #235: Plugin works on iOS but not on Android

ronneylogs, 2023-04-13:
My code works perfectly fine on iOS, but has not worked properly on Android. On Android, I am able to detect BLE devices, but upon connection, I am served the wrong BLE service. Does anyone have any idea why this is happening? Is this a known issue?

whatdtech, 2023-04-15:
```
    <!-- Allow Bluetooth -->
    <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />
    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />
    <!-- Needed only if your app looks for Bluetooth devices.
        If your app doesn't use Bluetooth scan results to derive physical
        location information, you can strongly assert that your app
        doesn't derive physical location. -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <!-- Needed only if your app makes the device discoverable to Bluetooth
         devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
    <!-- Needed only if your app communicates with already-paired Bluetooth
      devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <!-- If your application supports a service and can run on Android 10 (API level 29) or Android 11, you must also declare 
    the ACCESS_BACKGROUND_LOCATION permission to discover Bluetooth devices. -->
    <!-- <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" /> -->

  List<Permission> permissions = [
        Permission.location,
        Permission.storage,
        Permission.bluetooth,
        Permission.bluetoothConnect,
        Permission.bluetoothScan,
        ];
   await permissions.request();

    <!-- Allow Bluetooth -->
    <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />
    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />
    <!-- Needed only if your app looks for Bluetooth devices.
        If your app doesn't use Bluetooth scan results to derive physical
        location information, you can strongly assert that your app
        doesn't derive physical location. -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <!-- Needed only if your app makes the device discoverable to Bluetooth
         devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
    <!-- Needed only if your app communicates with already-paired Bluetooth
      devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <!-- If your application supports a service and can run on Android 10 (API level 29) or Android 11, you must also declare 
    the ACCESS_BACKGROUND_LOCATION permission to discover Bluetooth devices. -->
    <!-- <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" /> -->

  List<Permission> permissions = [
        Permission.location,
        Permission.storage,
        Permission.bluetooth,
        Permission.bluetoothConnect,
        Permission.bluetoothScan,
        ];
   await permissions.request();

```
All the above permissions in xml file and asking permissions during app initialisation are mandatory for Android latest versions.

Just do flutter clean and then build. It works for sure on all android versions

ronneylogs, 2023-04-16:
Did you use a permission handler plugin for Flutter or you can do that without? Also where in your code do you put your permissions.request line?

ronneylogs, 2023-04-27:
Wanted to give an update on this. On the Android side I was served three different services while on the iOS I was only served the service that was expected. It is unknown why this happened, but I also realized I can filter out the unwanted services by providing the UUID in the code.  This is a workaround, not a real solution.

closed 2023-04-27 by ronneylogs