chipweinberger/flutter_blue_plus #236: [Question] 19 dependencies seems excessive. Are all really needed?

chipweinberger, 2023-04-16:
Flutter Blue Plus uses 19 dependencies. 

As we all know, more dependencies == more maintenance & things breaking more often

Can you provide guidance on what these are used for? It would be great to get rid of any unneeded dependencies.

I would be happy to help.

```
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_flutter_app % flutter pub add flutter_blue_plus
Resolving dependencies...
  async 2.10.0 (2.11.0 available)
  characters 1.2.1 (1.3.0 available)
  collection 1.17.0 (1.17.1 available)
+ convert 3.1.1
+ fixnum 1.1.0
+ flutter_blue_plus 1.4.0
  intl 0.17.0 (0.18.0 available)
  js 0.6.5 (0.6.7 available)
  matcher 0.12.13 (0.12.15 available)
  material_color_utilities 0.2.0 (0.3.0 available)
  meta 1.8.0 (1.9.1 available)
  path 1.8.2 (1.8.3 available)
  path_provider_android 2.0.24 (2.0.25 available)
  path_provider_foundation 2.2.1 (2.2.2 available)
+ protobuf 2.1.0
+ rxdart 0.27.7
  source_span 1.9.1 (1.10.0 available)
  test_api 0.4.16 (0.5.1 available)
  win32 3.1.4 (4.1.3 available)
Changed 5 dependencies!
```

chipweinberger, 2023-04-16:
My findings:

Could easily be removed (in order of ease):
**meta**: used for a single _@immutable_ annotation for `CharacteristicProperties`
**rxdart**: used in a single place for `BehaviorSubject`
**convert**: used only to encode & decode a few hex strings
**collection**: only used for `compareAsciiLowerCase` and `ListEquality` to get a hash of a list

Required:
**flutter**: needed for `MethodChannel` and `EventChannel`, which are expected for native extension channels.

Possibly Required:
**protobuf**: Of course, we will always want protobuf tooling to do the codegen. It's a great tool for this use case. But do we need it as a runtime dependency? Possibly not. I think if we generated protobuf code with the JSON option, then the generated could would free-standing, only using standard libraries like `org.json` in Java, `JSONSerialization` in Swift, and `jsonDecode` in Dart.  But I am not sure. Would be awesome though to remove such a complicated dependency.

chipweinberger, 2023-04-16:
For this schema:
```
message MtuSizeRequest {
  string remote_id = 1;
  uint32 mtu = 2;
}
```

protobuf:

```
// Generated by the protocol buffer compiler. DO NOT EDIT!
// source: mtu_size_request.proto

import 'dart:core' show int, bool, double, String, List, override;
import 'package:protobuf/protobuf.dart';

class MtuSizeRequest extends GeneratedMessage {
  static final BuilderInfo _i = BuilderInfo('MtuSizeRequest')
    ..aOS(1, 'remoteId')
    ..a<core.int>(2, 'mtu', PbFieldType.OU3)
    ..hasRequiredFields = false;

  MtuSizeRequest() : super();
  MtuSizeRequest.fromBuffer(List<int> i,
      [ExtensionRegistry r = ExtensionRegistry.EMPTY])
      : super.fromBuffer(i, r);
  MtuSizeRequest.fromJson(String i,
      [ExtensionRegistry r = ExtensionRegistry.EMPTY])
      : super.fromJson(i, r);
  MtuSizeRequest clone() => MtuSizeRequest()..mergeFromMessage(this);
  BuilderInfo get info_ => _i;
  static MtuSizeRequest create() => MtuSizeRequest();
  static PbList<MtuSizeRequest> createRepeated() =>
      PbList<MtuSizeRequest>();
  static MtuSizeRequest getDefault() {
    if (_defaultInstance == null) _defaultInstance = MtuSizeRequest._();
    return _defaultInstance;
  }

  static MtuSizeRequest _defaultInstance;
  static void $checkItem(MtuSizeRequest v) {
    if (v is! MtuSizeRequest) checkItemFailed(v, 'MtuSizeRequest');
  }

  String get remoteId => $_getS(0, '');
  set remoteId(String v) {
    $_setString(0, v);
  }

  bool hasRemoteId() => $_has(0);
  void clearRemoteId() => clearField(1);

  int get mtu => $_get(1, 0);
  set mtu(int v) {
    $_setUnsignedInt32(1, v);
  }

  bool hasMtu() => $_has(1);
  void clearMtu() => clearField(2);
}

```

OpenAPI:

```
import 'package:json_annotation/json_annotation.dart';

part 'mtu_size_request.g.dart';

@JsonSerializable()
class MtuSizeRequest {
  final String remote_id;
  final int mtu;

  MtuSizeRequest({required this.remote_id, required this.mtu});

  factory MtuSizeRequest.fromJson(Map<String, dynamic> json) =>
      _$MtuSizeRequestFromJson(json);

  Map<String, dynamic> toJson() => _$MtuSizeRequestToJson(this);
}
```

chipweinberger, 2023-04-27:
closing. these were removed.

closed 2023-04-27 by chipweinberger

EmreSURK, 2023-07-29:
Protobuf may use some native code on some platforms. This may create build issues and it can be a blocker. Please keep in mind this important deal break while having a decision. Thanks.

chipweinberger, 2023-07-29:
protobuf was removed for this reason :)