chipweinberger/flutter_blue_plus #242: after setNotifyValue(true) characteristic.write exceptions throws exceptions

MrCsabaToth, 2023-04-20:
In the past the exception was PlatformException (#68) and I could catch it. Some time ago, but with the latest beta for sure (`3.10.0-1.1.pre`) the exception is not even a `PlatformException`, but a "silent" plain Exception which I can only catch with a `runZonedGuarded` based "pacman" exception catcher at the main level.
This happens with several of my users in release version and I don't know so far what to do about it. What is also interesting is that sometimes it seems like that the Bluetooth devices actually receive the data packets despite of the exception.
This scenario is common when a characteristic is both readable, writeable and also able to stream (setNotifyValue(true)). It only happens if I write to the same charactersitic which I already listen to with setNotifyValue(true). It doesn't happen in debug version.

chipweinberger, 2023-04-27:
Please lmk if you find a fix and can open a PR.

chipweinberger, 2023-04-28:
What is the stack trace?

MrCsabaToth, 2023-04-28:
This is how the crash looks in the Dart/Flutter layer (logged by special logging developed into my app for dealing with crashes happening with released apps. I don't let these crashes crash the whole app - just because something is wrong with a device for example - so I don't have a native call stack in the Play Store):
```
{DATA_HANDLER}  {_executeControlOperationCore}  {controlPoint.write; Exception: Failed to write the characteristic; 
#0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:135)
#1      _rootRunUnary (dart:async/zone.dart:1407)
<asynchronous suspension>
#2      KayakFirstDescriptor._executeControlOperationCore (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:133)
<asynchronous suspension>
#3      KayakFirstDescriptor.configureDisplay (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:213)
<asynchronous suspension>
#4      KayakFirstDescriptor.postPumpStart (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:237)
<asynchronous suspension>
#5      FitnessEquipment.postPumpStart (package:track_my_indoor_exercise/devices/gadgets/fitness_equipment.dart:521)
<asynchronous suspension>
#6      RecordingState._connectOnDemand (package:track_my_indoor_exercise/ui/recording.dart:278)
<asynchronous suspension>
}  {18 April 2023 07:37:04.023 PM}  {ERROR}
```

This is the command which induces the crash (the `setNotifyValue(true)` was called on the same characteristic ~5-10 seconds before): https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/8acaf9ff09047c815dc30b0b41cbf4859e8a0c2c/lib/devices/device_descriptors/kayak_first_descriptor.dart#L134

What is stunning, is that this command is surrounded with a "pacman" style exception catcher (which catches generic exception). Some time ago I used `PlatformException` but I noticed that it didn't catch the exception. But despite of the pacman catch, the crash somehow escapes, and the only thing which catches it is this `runZonedGuarded` construct at the main level: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/8acaf9ff09047c815dc30b0b41cbf4859e8a0c2c/lib/main.dart#L18

Honestly I cannot fully understand how the exception can slip through the Exception catch and end up here in the `main`. Maybe the `async` execution matters, but still...

MrCsabaToth, 2023-04-28:
I'll try to work on reproducing it in my development environment to obtain a LogCat.

chipweinberger, 2023-04-28:
> But despite of the pacman catch, the crash somehow escapes

I think maybe because the exception is being thrown in the stream `then` clause? Maybe `throw` should be moved to the outer layer.

```
    var response = FlutterBluePlus.instance._methodStream
        .where((m) => m.method == "WriteCharacteristicResponse")
        .map((m) => m.arguments)
        .map((buffer) => protos.WriteCharacteristicResponse.fromBuffer(buffer))
        .where((p) =>
            (p.request.remoteId == request.remoteId) &&
            (p.request.characteristicUuid == request.characteristicUuid) &&
            (p.request.serviceUuid == request.serviceUuid))
        .first
        .then((w) => w.success)
        .then((success) => (!success)
            ? throw Exception('Failed to write the characteristic')
            : null)
        .then((_) => null);
```

chipweinberger, 2023-04-28:
ChatGPT confirms. Damn, this FlutterBlue codebase is pretty crap! And then then 'readCharacteristic' function does not have any error handling.... so crappy.

[image]

closed 2023-04-28 by chipweinberger