chipweinberger/flutter_blue_plus #244: Exception: Another scan is already in progress

roscadalexandru, 2023-04-20:
It seems that after the scan timeout passes the scan doesn't actually stops. I tried with stopScan as well but same error.

Is there any solution for this?

chipweinberger, 2023-04-27:
Please re-pull master. And open a PR if you can fix it.

chipweinberger, 2023-04-28:
try:

```
      if ( _flutterBlue.isScanningNow == false) {
        await _flutterBlue.startScan();
     }
```

clemens-, 2023-05-07:
I have the same issue. I tried the following:
`Future<void> startScan() async {
    print("Starting scan...");
    if ( _flutterBlue.isScanning == true) {
      print("STILL SCANNING, STOPPING...");
      _flutterBlue.stopScan();
    }
    await _flutterBlue.startScan(withServices: [Guid(serviceEasyoilerUuid)], timeout: const Duration(seconds: 4));
  }`

Output:
`I/flutter (13341): Starting scan...
E/flutter (13341): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: Another scan is already in progress.
E/flutter (13341): #0      FlutterBluePlus.scan (package:flutter_blue_plus/src/flutter_blue_plus.dart:144:7)
E/flutter (13341): <asynchronous suspension>`

It seems that the isScanning-Boolean is not working correctly.

clemens-, 2023-05-07:
I fixed it by calling stopScan before restarting the scan. My inkwell-widget loks like this:

```dart
return InkWell(
                                          splashColor: Colors.transparent,
                                          focusColor: Colors.transparent,
                                          hoverColor: Colors.transparent,
                                          highlightColor: Colors.transparent,
                                          onTap: () {
                                            setState(() {
                                              Provider.of<BleService>(context, listen: false).stopScan();
                                              Provider.of<BleService>(context, listen: false).startScan();
                                            });
                                            logFirebaseEvent('CONNECT_PAGE_Container_okbya106_ON_TAP');
                                            logFirebaseEvent('Container_navigate_to');
                                          },
                                          child: _refreshWidget(context),
                                        );
```

chipweinberger, 2023-05-08:
@clemens- , `isScanning` is not a bool. It is a `Stream<bool>`. `isScanningNow` is the bool (master branch).

chipweinberger, 2023-07-20:
closing. please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger