chipweinberger/flutter_blue_plus #245: Future of this library

Indigi-managed, 2023-04-23:
Hey @boskokg 

We really love this package and it's helped a lot of people out so I'm posting this here to ask, how can we as a community better help you?

Some ideas:

- Add another trusted owner/contributor to approve pull-requests
- Add a sponsor button to allow others to sponsor this repo (I would happily pay monthly for the future of this package)

Keen to hear your thoughts.

chipweinberger, 2023-04-27:
I'm using this library for a new project, so I'll be working on it for the next while.

boskokg, 2023-04-28:
@chipweinberger can you please send me the email that I can use to add permission for uploading new version to pub.dev

closed 2023-04-28 by boskokg

chipweinberger, 2023-05-05:
@boskokg were you able to add me?

1. you need to create a pub.dev publisher
2. add me to the publisher
3. add the package to the publisher

boskokg, 2023-05-06:
Hello @chipweinberger 
I have sent the invitation

[image]

chipweinberger, 2023-05-06:
great! got it!