chipweinberger/flutter_blue_plus #264: Major Design Issues with Flutter Blue

chipweinberger, 2023-04-29:
I've been working on Flutter Blue Plus for a couple days now, and as I'm looking at the issues, it's clear that there are some major design issues with Flutter Blue.

### Timing

Flutter Blue has some obvious timing problems.  See commits like this for reference: https://github.com/boskokg/flutter_blue_plus/commit/a85378b8a5eb9178bdeda9b278947489fbe03c18

Before this commit we would:
1. invoke a platform method
2. Wait for the first result

After this commit: 
1. start waiting for the result
2. invoke a platform method

This fixes a big problem. But causes a new one, what if multiple callers call `read` at the same time? They'll all do step **1**, and then start waiting on step **2**. But which result will they get? The design of FlutterBlue with a shared channel for  these methods means its very possible for later callers to accidentally return the data meant for earlier callers. 

```
  Future<void> write(List<int> value, {bool withoutResponse = false}) async {

   ...

      var responseStream = FlutterBluePlus.instance._methodStream
        .where((m) => m.method == "WriteCharacteristicResponse")
        .map((m) => m.arguments)
        .map((buffer) => protos.WriteCharacteristicResponse.fromBuffer(buffer))
        .where((p) =>
            (p.request.remoteId == request.remoteId) &&
            (p.request.characteristicUuid == request.characteristicUuid) &&
            (p.request.serviceUuid == request.serviceUuid));

      // Start listening now to ensure we don't miss the response
      Future<protos.WriteCharacteristicResponse> futureResponse = responseStream.first;

      await FlutterBluePlus.instance._channel
        .invokeMethod('writeCharacteristic', request.writeToBuffer());

      // wait for response, so that we can check for success
      protos.WriteCharacteristicResponse response = await futureResponse;
      if (!response.success) {
        throw Exception('Failed to write the characteristic');
      }

      return Future.value();
}
```

### Error Handling

Overall, Flutter Blue was not designed with error handling front-of-mind.

This is clear with issues like https://github.com/boskokg/flutter_blue_plus/issues/242, where its super easy to miss an exception because Flutter Blue code was just not written correctly. This bug has been fixed, but it makes it clear that Error Handling was not a major consideration, or strongly tested, during development.

### General Sloppiness

This is harder to write down, but there is lots of general sloppiness & inconsistency in the code. Duplicate variables in the Dart API. Try/catch clauses that just log in native code. Code which was needlessly hard to read. etc. 

### flutter reactive blue

Do to these issues, and after reading the [flutter_reactive_blue](https://github.com/PhilipsHue/flutter_reactive_ble) codebase, I am going to be switching my efforts to that repository. 

1. they do not have the timing issue
2. all functions have error handling considered
3. the code itself is clear and consistent

```
flutterReactiveBle.scanForDevices(withServices: [serviceId], scanMode: ScanMode.lowLatency).listen((device) {
      //code for handling results
    }, onError: () {
      //code for handling error
    });
```

### Covering my own ass

If flutter_reactive_blue does not work out, there is of course still a chance I will return to this codebase. I'll give myself that out =)

I would also consider writing my own ble driver instead.

chipweinberger, 2023-04-29:
I've updated this recommendation on the readme. https://github.com/boskokg/flutter_blue_plus/commit/32c7112ae3b325d695aa1f23516e764c1b43b5fd

Hopefully I'm not pissing people off. Just trying to help people write the best apps they can!

chipweinberger, 2023-05-01:
So I am running into issues with `flutter_reactive_ble` as well 😩

https://github.com/PhilipsHue/flutter_reactive_ble/issues/728
https://github.com/PhilipsHue/flutter_reactive_ble/issues/385

So, I am going to start tackling these design issues in `flutter_blue_plus`.

This simplest way to fix **Design Issue 1: Timing**, is to just wrap `read` & `write` in a mutex, so that simultaneous calls to read & write will be serialized.

**Design Issue 2: Error Handling** & **Design Issue 3: Inconsistency**, I'll just tackle as-needed.

chipweinberger, 2023-05-02:
Fixed Design Issue 1: https://github.com/boskokg/flutter_blue_plus/commit/27a2a445968b07c9799c40872397af1566478e68

Sajonji, 2023-05-02:
Great to see progress on this library again !

@chipweinberger: do you have any idea if and, in case of yes, when you can update the pub dev packages with your changes ?

Nolence, 2023-05-02:
Our team has experienced challenges with both flutter_blue(_plus) and flutter_reactive_ble. While flutter_blue_plus offers a more understandable codebase, its capabilities are limited and it has been poorly kept up with in the past. On the other hand, flutter_reactive_ble is reliable for developing real-time and robust applications. I really like their API and the stream-based implementations were easy to use. However, some long-standing issues have remained unaddressed, which is understandable considering it's an open-source project.

Recently, I've been exploring [flutter_btleplug](https://github.com/trobanga/flutter_btleplug) as a potential final Bluetooth solution for our projects. Although still heavily under development, the necessary tooling is already in place, and the parent project [btleplug](https://github.com/deviceplug/btleplug) has a large user base and a good maintainer.

Just some random thoughts I had lol sorry

chipweinberger, 2023-05-02:
@Sajonji I am still waiting on access to pub.dev. I don't plan to update it right away. I need to finish integrating & testing BLE in my app to make sure the changes work. Feel free to pull master and let me know if you have issues. 

@Nolence I totally agree. flutter_reactive_ble has way too many classes & unnecessary abstraction. The Swift code is incomprehensible. https://github.com/trobanga/flutter_btleplug looks interesting. But needing a rust toolchain is way too much complexity IMO. 

I like the simplicity of flutter_blue, which is why I just plan to make improvements as needed.

> Our team has experienced challenges with both flutter_blue(_plus) and flutter_reactive_ble. 

Anything specific?

Nolence, 2023-05-08:
Yeah, our issues have mostly come from connecting and disconnecting reliably (repeatedly) with flutter_blue(_plus). Another issue we ran into was auto fragmentation of packets. In flutter_blue that's mostly taken care of I think. In flutter_reactive_ble I don't think that happens based on some tests I ran on an older iphone and android device. Flutter_reactive_ble also doesn't have support for manually messing with a characteristic's descriptors which we would like for some odd pieces of hardware we have.

Sajonji, 2023-05-09:
@chipweinberger 

> Feel free to pull master and let me know if you have issues. I have none so far.

I did pull master. My Android Studio however isn't able to correctly work with any of the android part of the library. Eg FlutterBluePlusPlugin.java shows me 384 errors right now, because Android Studio can't resolve any of the androidx and flutter imports. 
I'm only a Junior right now, any hint on how to solve this issues (or use a flutter lib outside of pub dev) would be welcome. Internet searches turned up nothing so far.

Sajonji, 2023-05-09:
@chipweinberger 

> Feel free to pull master and let me know if you have issues. I have none so far.

Yes, i did pull master and changed the dependency in my flutter project from the pub dev FlutterBluePlus to my local clone. My flutter app builds and runs fine.

However, when i try to open the checked out code in Android Studio to implement some changes, Android Studio isn't able to resolve anything. I get 981 errors in FlutterBluePlusPlugin.java.
Do you have any hint how i need to configure Android Studio with the pulled code ?

closed 2023-05-22 by chipweinberger