chipweinberger/flutter_blue_plus #268: android turnBluetoothOn uses deprecated function and won't work with Android version 33

Sajonji, 2023-05-02:
in flutter_blue_plus/blob/master/android/src/main/java/com/boskokg/flutter_blue_plus/FlutterBluePlusPlugin.java in line 250 :
`case "turnOn":
      {
        if (!mBluetoothAdapter.isEnabled()) {
          result.success(mBluetoothAdapter.enable());
        }
        break;
      }`

BluetoothAdapter.enable() usage is discouraged for a few versions already, and is now hard deprecated and not working anymore in version 33 (Android 13): [documentation](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#enable())
Instead an Intent with ACTION_REQUEST_ENABLE should be used to let the user decide to turn Bluetooth on.

Same issue for turnOff with BluetoothAdapter.disable()

rasheed-ct, 2023-05-02:
i have same problem in android 13 
it's throw this  ` checkPermission: android.permission.BLUETOOTH_ADMIN, pid=30012, uid=10414, ret=true`

chipweinberger, 2023-05-05:
please open a PR

Sajonji, 2023-05-10:
I'll try to do a pull request soon.
This should definitely be changed, since it doesn't work anymore beginning from android 13.
i've got a mail about a comment saying "it was just deprecated and will continue to work", but i cannot see that comment here. However, this is factually wrong. To quote the documentation:

> Starting with [Build.VERSION_CODES.TIRAMISU](https://developer.android.com/reference/android/os/Build.VERSION_CODES#TIRAMISU), applications are not allowed to enable/disable Bluetooth. Compatibility Note: For applications targeting [Build.VERSION_CODES.TIRAMISU](https://developer.android.com/reference/android/os/Build.VERSION_CODES#TIRAMISU) or above, this API will always fail and return false. If apps are targeting an older SDK ([Build.VERSION_CODES.S](https://developer.android.com/reference/android/os/Build.VERSION_CODES#S) or below), they can continue to use this API.

Sajonji, 2023-05-19:
fixed in pull request #282

closed 2023-05-19 by Sajonji

SunlightBro, 2023-08-02:
Related to this,
**turnOff** does also not work on Android 33.

So something similar to the changes for `turnOn` are probably also needed for `turnOff` 
(just pseudo code, I have not much experience with java)

```java
  case "turnOff":
  {
    if (mBluetoothAdapter.isEnabled()) {
      Intent disableBtIntent = new Intent(BluetoothAdapter.ACTION_REQUEST_DISABLE);
      activityBinding.getActivity().startActivityForResult(disableBtIntent, disableBluetoothRequestCode);
      result.success(true);
    } else {
      result.success(false);
    }
    break;
  }
```

@chipweinberger let me know if you want a separat issue, or a pull request

chipweinberger, 2023-08-02:
yes please open an issue

SunlightBro, 2023-08-08:
turnOn() on Android 13, crashes if the "Nearby Devices" is not yet given

```log
java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{ea340e8 8825:com.distolib.example/u0a403} (pid=8825, uid=10403) requires android.permission.BLUETOOTH_CONNECT
```

Can flutter_blue_plus first check/ask for the permission before starting the Intent?

chipweinberger, 2023-08-08:
> Can flutter_blue_plus first check/ask for the permission before starting the Intent?

We should.

chipweinberger, 2023-08-08:
merged: 4f043bafddae9c7990ec05940924a21bbe0efd87