chipweinberger/flutter_blue_plus #280: Permission requirements for BLE scanning on Android 12+ are too pedantic

virshe00, 2023-05-11:
The permission checks before BLE scanning on Android 12+ are too pedantic/aggressive.

[This check](https://github.com/boskokg/flutter_blue_plus/blob/d562274902f1d0bdc7f08b9e42f2d1452380063d/android/src/main/java/com/boskokg/flutter_blue_plus/FlutterBluePlusPlugin.java#LL272C28-L272C28) checks for fine location before attempting to scan. This check is incorrect.

According to the [Android specification](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.app.PendingIntent)), fine location is not required when `usesPermissionFlags="neverForLocation"` is specified in the app manifest.

itamardo, 2023-05-23:
I prefer not to ask my users for fine location on android 12 so as long as this is forced in 1.5.2, I forked and 
changed AndroidManifest.xml to:

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  xmlns:tools="http://schemas.android.com/tools"
  package="com.boskokg.flutter_blue_plus">
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"
        tools:targetApi="s" />
</manifest>
```

and changed FlutterBluePlusPlugin.java in the scan code to omit the check of fine location for type "s" like this:
```
case "startScan":
      {
        ensurePermissionBeforeAction(Build.VERSION.SDK_INT >= Build.VERSION_CODES.S ? Manifest.permission.BLUETOOTH_SCAN : Manifest.permission.ACCESS_FINE_LOCATION, (grantedScan, permissionScan) -> {
          if (grantedScan) {
            ensurePermissionBeforeAction(Build.VERSION.SDK_INT >= Build.VERSION_CODES.S ? Manifest.permission.BLUETOOTH_CONNECT : null, (grantedConnect, permissionConnect) -> {
              if (grantedConnect)
                startScan(call, result);
              else
                result.error(
                        "no_permissions", String.format("flutter_blue plugin requires %s for scanning", permissionConnect), null);
            });
          }
          else
            result.error(
                    "no_permissions", String.format("flutter_blue plugin requires %s for scanning", permissionScan), null);
        });
        break;
      }
```

The changelog for 1.5.0 states that the fine location requirement was added on purpose on Android12. I never encountered the need myself on an Android12 phone and the documentation clearly states that fine location isn't needed for scanning.
Can anyone share a phone model that needs fine location with Android 12?

virshe00, 2023-05-30:
> I prefer not to ask my users for fine location on android 12 so as long as this is forced in 1.5.2, I forked and changed AndroidManifest.xml to:
>
> ```
> <manifest [...]

If phones are using Android 12, you'd expect them all to have the same permission requirements wouldn't you?

itamardo, 2023-05-31:
@virshe00 I would, yes, which is why I don't understand the decision made in 1.5.0 and why I feel ok with the fork I made. Still - the changelog hints that some people faced Android 12 phones that failed to work with the documented requirements, so maybe some manufacturer did something wrong? idk. Myself, I never encountered an Android 12 phone that needed fine location for BLE, but I didn't test more than a few manufacturers

virshe00, 2023-05-31:
@itamardo Yeah, that makes sense. I can't find any reference to anyone having issues and if a small phone manufacturer isn't the Android specification then idk what else they're doing and they sound like a bad brand.

Regardless, I think the core of the issue comes from requiring fine location permissions, but only if you specify `neverForLocation`.

@chipweinberger Could you please make a decision as to what you think the course of action should be? I can understand why you might want to enforce precise location. *If* there's no way to detect whether the manifest specifies `neverForLocation`, it *might* be safer to simply enforce fine location. However, I think reverting [the commit](https://github.com/boskokg/flutter_blue_plus/commit/d562274902f1d0bdc7f08b9e42f2d1452380063d) a note/section in the README might be sufficient.

xbug42, 2023-06-01:
> Can anyone share a phone model that needs fine location with Android 12?

I would be interested too... This new location permission request is very annoying and against the Android specs (as long as BT is not used for location, obviously)

GorIvanov, 2023-06-09:
I would like to contact flutter developers. I hope you read github.
You are creating a cross-platform framework that should behave the same on all platforms and have common interaction mechanisms. Now there are more than 10 options for plugins for working with bluetooth on pub.dev. I believe that you, as developers of a cross-platform framework, are obliged to develop and make part of the framework all the basic mechanisms for working with such things as bluetooth, cameras, wifi, and the like, as is done for example in QT.

itamardo, 2023-06-15:
My manifest above had a missing line that was apparently needed for Android 11 phone so I updated it like this:

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  xmlns:tools="http://schemas.android.com/tools"
  package="com.boskokg.flutter_blue_plus">
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"
        tools:targetApi="s" />
</manifest>```

virshe00, 2023-06-15:
> My manifest above had a missing line (fine location for sdk <= 30) that was apparently needed for an Android 11 phone so I updated it like this:
> 
> ```
> <manifest xmlns:android="http://schemas.android.com/apk/res/android"
>   xmlns:tools="http://schemas.android.com/tools"
>   package="com.boskokg.flutter_blue_plus">
>     <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
>     <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
>     <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="30" />
>     <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
>     <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
>     <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"
>         tools:targetApi="s" />
> </manifest>
> ```

I wondered about this, thanks for pointing it out.
I'll try and get this thing done over the weekend.

chipweinberger, 2023-07-20:
a lot of this changed in 1.7.0+

please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger