chipweinberger/flutter_blue_plus #283: Scanning gets no readings for 32 seconds and then gets readings for one startscan cycle and it repeats

Daniel-Harding, 2023-05-13:
It might be that this issue is not a Flutter Blue Plus issue and maybe my code.  I am seeing when I open a 4 second triple scan cycle (long for channel 37 (3.9 seconds) and short for channel 38 and channel 39 (about 10mSec each) ... so really three separate scan cycles  I receive readings on the 4th cycle and 28th cycle and 52 cycle (increments of 24/3 =8 ??).... so about every 32 seconds I get a reading from a device that is sending out an advertising beacon on channel 37 only at 0.33 Second intervals (from an embedded device that I have verified is transmitting with Nordic Semi's NRF connect cellphone application ... so I am only seeing about 2% of the total beacons being transmitted... just wondering if anyone else has seen an issue like this and if there are any settings that would mask or block readings for a time interval after receiving the first one...?  Ideas?

chipweinberger, 2023-05-19:
please open a PR if you fix the problem

Daniel-Harding, 2023-06-28:
So it turns out that Android restricts the application to 5 scans per 30 second interval.  So our attempt to have one scan and two quick scans would cause the OS to block the additional scan requests and thus the long periods where we did not receive Bluetooth advertisements from our sensor.   Also the StartScan method uses a stream that caused us some issues because we were re-reading old readings as new readings (repeats)....so maybe we didn't understand how to use the stream correctly.... but ultimately we switched to use Scan instead of StartScan and now it works well.  One final point... regarding Android when you use Low Latency mode and then declare a scan it starts on channel 37 for 4.096 seconds and then switches to channel 38 for 4.096 seconds and then channel 39 for 4.096 seconds and then back to channel 37.  So if you started and stopped 5 scans that were each 4.096 seconds long you could get about 20.5 seconds of nearly continuous scanning on channel 37 BUT then you would have to wait 9.5 seconds before the OS would allow you to scan again... so not very helpful if you want to scan on just one channel.  iOS scans for 30mSecs and then stops its scan for 10 mSecs and this is controlled by iOS and appears to not be configurable in any way by the users application.... so Apple recommends a beacon interval of one of the following 152.5ms,211,25ms,318,75ms,417.5ms,546.25ms,760ms,852.5ms,1022.5ms,1285ms

chipweinberger, 2023-07-20:
glad that you figured it out.

To summarize, scanning is rate limited.

closed 2023-07-20 by chipweinberger