chipweinberger/flutter_blue_plus #284: Queued data not showing.

corteXnzt, 2023-05-13:
I created an app based on this code. Using stream to receive new notifications on data.

https://github.com/0015/ThatProject/blob/master/Esp32_dust_sensor_ble_FLUTTER/flutter_app_esp32_dust_sensor/lib/sensor_page.dart

My ble device send successive ble notifications using the nordic uart service, its working on an android app based on java but I'm creating a new one that is flutter based. I think the notifications are very fast that it was not able to process the first 3 data sent. 

1. Do you think this is an issue with the library?
[image]

2. If this is not an issue with the library is their an example on how to implement subscribing the values without using stream?

TIA

chipweinberger, 2023-05-19:
please open a PR if you fix the problem

chipweinberger, 2023-07-20:
please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger