chipweinberger/flutter_blue_plus #285: autoConnect doesn't work if BT was turned off then on again

zemmyang, 2023-05-15:
Scenario: 
- app opens while BT is on
- app reads data from device
- user turns off BT (for example, turned on airplane mode)
- user manually turns on BT
- what happens: app doesn't reconnect to device

---

I tried using RxDart's RetryWhenStream to force the reconnection:

```
  RetryWhenStream(
        () => flutterBluePlus.state
            .map((i) async {

              if (i == BluetoothState.on){
                await device.connect(autoConnect: true);
                return dataStream;
              }
              else {
                throw "BT not on";
              }
        }),
          (Object e, StackTrace s) =>
          Rx.timer<void>(null, const Duration(seconds: 1)),
  ).listen((event) {
// stuff
  });
```

where the [initial setup for the stream is done like this](https://github.com/boskokg/flutter_blue_plus/issues/274#issuecomment-1539803590). i also tried putting that entire function in the if BT.on block in the code above, but it still doesn't reconnect.

Maybe I'm missing something?

chipweinberger, 2023-05-19:
please open a PR if you fix the problem

chipweinberger, 2023-07-20:
please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger