chipweinberger/flutter_blue_plus #286: iOS 16: First time `FlutterBluePlus.instance.isOn` goes to False, even though Bluetooth is enabled

ma2yama, 2023-05-16:
I'm calling `isOn` to see if Bluetooth is enabled, but the first time `isOn` goes to False, even though Bluetooth is enabled.
Is there any way to solve this problem?

chipweinberger, 2023-05-19:
please open a PR if you fix the problem

baterdene-ums, 2023-05-30:
I had the same problem

bataa07, 2023-05-30:
I had the same problem

ma2yama, 2023-06-07:
I worked around the problem by sandwiching sleep as follows:

```dart
Future<bool> blueIsOn() async {
  final isOn = await _flutterBlue.isOn;
  if (isOn) return true;

  sleep(const Duration(milliseconds: 100));
  return await _flutterBlue.isOn;
}
```

chipweinberger, 2023-07-20:
please try 1.7.6+, reopen a new issue if you still have problems.

meynety, 2023-08-01:
Still having the same issue with `1.12.8` !

chipweinberger, 2023-08-01:
`isOn` is deprecated in 1.12.9

so try master branch

```
dependencies:
  flutter_blue_plus:
    git:
      url: https://github.com/boskokg/flutter_blue_plus.git
      ref: master
```

chipweinberger, 2023-08-01:
I cannot repro in 1.12.9 on macOS using the example. **note**: adapter_state 4 == on

```
flutter: [FBP] <setLogLevel> args: 5
flutter: [FBP] <stopScan> args: null
2023-08-01 15:45:37.197 macos[18831:1683771] [FBP-iOS] handleMethodCall: setLogLevel
2023-08-01 15:45:37.197 macos[18831:1683771] [FBP-iOS] initializing CBCentralManager
2023-08-01 15:45:37.198 macos[18831:1683771] [FBP-iOS] handleMethodCall: stopScan
flutter: [FBP] <getAdapterState> args: null
2023-08-01 15:45:37.248 macos[18831:1683771] [FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
2023-08-01 15:45:37.248 macos[18831:1683771] [FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] <setLogLevel> result: 1
flutter: [FBP] <stopScan> result: 1
flutter: [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 4}
flutter: [FBP] <getAdapterState> result: {adapter_state: 4}
```

chipweinberger, 2023-08-01:
should be fixed in 1.12.9

reopen if you still have issues.

meynety, 2023-08-03:
@chipweinberger issue still present with `1.12.9`
> @meynety are you making sure isAvailable is true first?
I didn't, I do now, and isAvailable is `false`, which is not correct.

Here is log of  calling`isAvailable` and then `await FlutterBluePlus.adapterState.first `(sorry for horrible format, not sure why, it does this only for FBP logs...) : 
```
[FBP-iOS] handleMethodCall: setLogLevel
[FBP-iOS] initializing CBCentralManager
[FBP-iOS] handleMethodCall: setLogLevel
[FBP-iOS] handleMethodCall: isAvailable
flutter: [FBP] \^[[1;30m<isAvailable>\^[[0m result: \^[[1;33mfal<…>
flutter: Bluetooth available : false
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m result: \^[[1;33m{adapter_state: <…>
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m result: \^[[1;33m{adapter_state: <…>
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m args: \^[[1;35mnu<…>
[FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m result: \^[[1;33m{adapter_state: <…>
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
flutter: [FBP] \^[[1;30m[[ OnAdapterStateChanged ]]\^[[0m result: \^[[1;33m{adapter_state: <…>
flutter: [FBP] \^[[1;30m<setLogLevel>\^[[0m args: \^[[1;35<…>
```

chipweinberger, 2023-08-03:
```
        else if ([@"isAvailable" isEqualToString:call.method])
        {
            result(@(self.centralManager != nil));
        }
```

I should probably simplify "isAvailable" to just check for nil ^^

And then users should just do:

`await FlutterBluePlus.adapterState.where((s) => s == BluetoothAdapterState.on).first;`

chipweinberger, 2023-08-03:
should be fixed in 1.12.10

closed 2023-08-03 by chipweinberger

meynety, 2023-08-03:
> is is expected for adapterState to be "unknown" or "off" when first initializing. This is why you should do
> 
> await FlutterBluePlus.adapterState.where((s) => s == BluetoothAdapterState.on).first;

Okay, I suggest you to add this in the documentation and the main Readme, this behaviour is non intuitive at all.

If I may : how are you supposed to only check for bluetooth status without waiting for it to be `on` ? (just like the the old deprecated `FlutterBluePlus#isOn()` method)

chipweinberger, 2023-08-03:
> If I may : how are you supposed to only check for bluetooth status without waiting for it to be on ? (just like the the old deprecated FlutterBluePlus#isOn() method)

check isOn:

`await adapterState.first == BluetoothAdapterState.on`

wait  for it to turn on:

`await FlutterBluePlus.adapterState.where((s) => s == BluetoothAdapterState.on).first;`

> Okay, I suggest you to add this in the documentation and the main Readme, this behaviour is not intuitive at all, as there is no "initialization" step mentioned anywhere.

yes thats a good point. The API was designed by other people. I'll add some more explanation.