chipweinberger/flutter_blue_plus #287: 不能连接上蓝牙  OPPO手机 133报错

dulifu, 2023-05-16:
设备连接中...
W/e.device_repai( 9299): Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
E/CheckPermission( 9299): _bluetooth code = 10 
E/CheckPermission( 9299): _bluetooth code = 10 
D/BluetoothGatt( 9299): connect() - device: 40:22:D8:6F:96:26, auto: false
D/BluetoothGatt( 9299): registerApp()
D/BluetoothGatt( 9299): registerApp() - UUID=43925c6a-85b4-42c5-b090-44261e2081d8
D/BluetoothGatt( 9299): onClientRegistered() - status=0 clientIf=6
E/CheckPermission( 9299): _bluetooth code = 10 
D/BluetoothGatt( 9299): onClientConnectionState() - status=133 clientIf=6 device=40:22:D8:6F:96:26
D/FlutterBluePlugin( 9299): [onConnectionStateChange] status: 133 newState: 0
D/BluetoothGatt( 9299): cancelOpen() - device: 40:22:D8:6F:96:26
E/CheckPermission( 9299): _bluetooth code = 10 
D/BluetoothGatt( 9299): close()
E/flutter ( 9299): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: TimeoutException after 0:00:18.000000: Failed to connect in time.
E/flutter ( 9299): #0      BluetoothDevice.connect.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_device.dart:51:11)
E/flutter ( 9299): #1      _rootRun (dart:async/zone.dart:1390:47)
E/flutter ( 9299): #2      _CustomZone.run (dart:async/zone.dart:1300:19)
E/flutter ( 9299): #3      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:865:34)
E/flutter ( 9299): #4      _rootRun (dart:async/zone.dart:1390:47)
E/flutter ( 9299): #5      _CustomZone.run (dart:async/zone.dart:1300:19)
E/flutter ( 9299): #6      _CustomZone.runGuarded (dart:async/zone.dart:1208:7)
E/flutter ( 9299): #7      _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1248:23)
E/flutter ( 9299): #8      _rootRun (dart:async/zone.dart:1398:13)
E/flutter ( 9299): #9      _CustomZone.run (dart:async/zone.dart:1300:19)
E/flutter ( 9299): #10     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:1232:23)
E/flutter ( 9299): #11     Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
E/flutter ( 9299): #12     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
E/flutter ( 9299): #13     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
E/flutter ( 9299): #14     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:192:26)
E/flutter ( 9299): 
D/BluetoothGatt( 9299): unregisterApp() - mClientIf=6

dulifu, 2023-05-18:
It may be because the storage permission is not opened, and there is no recurrence now.

dulifu, 2023-05-22:
133 A status error still occurs

dulifu, 2023-05-22:
The code for the connection is as follows:
```
connectAndDiscoverServices(BluetoothDevice device, ValueSetter<bool> action,
      {int seconds = 20}) async {
    _connected = false;
    _currentDevice = device;

    // removeNotifySubscription();
    // removeWriteSubscription();
    // removeStateSubscription();

    _stopConnectTimer();
    _connect_timer = Timer(Duration(seconds: seconds), () {
      action(false);
    });

    await _currentDevice.connect(
        autoConnect: false, timeout: Duration(seconds: seconds - 2));

    List<BluetoothService> services = await _currentDevice.discoverServices();
    ...
}
```

chipweinberger, 2023-05-22:
no idea. But If you fix the issue feel free to open a PR.

imobdev-git, 2023-06-17:
Hello, did you solved this issue ? I am getting same issue in Android. Below is the debuger log. 

`I/BluetoothAdapter(18738): STATE_ON
D/BluetoothGatt(18738): connect() - device: 4C8276_0, auto: true
D/BluetoothGatt(18738): registerApp()
D/BluetoothGatt(18738): registerApp() - UUID=31d5da4e-5112-4bd0-adfc-5777dbb2cbca
D/BluetoothGatt(18738): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt(18738): onClientConnectionState() - status=133 clientIf=6 device=4C8276_0
D/FlutterBluePlugin(18738): [onConnectionStateChange] status: 133 newState: 0
`

chipweinberger, 2023-07-20:
please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger