chipweinberger/flutter_blue_plus #289: startScan method parameter withServices problem

dulifu, 2023-05-18:
startScan method parameter question, normally can scan successfully, but after I add withServices parameter, can not scan the corresponding device, my method is as follows:
```
startScan(
      {int seconds = 10,
      Function(ScanResult) scanResultBlock,
      VoidCallback timeoutBlock}) {
    stopScan();
    _flutterBlue.startScan(
        timeout: Duration(seconds: seconds),
        withServices: [Guid("0000a002-0000-1000-8000-00805f9b34fb")]);
    _scan_timer = Timer(Duration(seconds: seconds), () {
      '搜索超时'.log();
      if (timeoutBlock != null) timeoutBlock();
    });

    /// 发现设备
    _scanScription = BluetoothManage().scanResults.listen((results) {
      for (ScanResult result in results) {
        if (scanResultBlock != null) {
          scanResultBlock(result);
        }
      }
    });
  }
```

I wonder if Service here refers to the UUID of the peripheral Service.

saguado69, 2023-05-18:
This works fine for me, using same device from IRXON ...

await ble.startScan(
              withServices: [Guid('0000ffe0-0000-1000-8000-00805f9b34fb')],
              timeout: const Duration(seconds: 4));
          try {
            // LISTEN FOR SCANNED BLE DEVICES
            ble.scanResults.listen((results) async {
              if (results.isNotEmpty) {
                for (ScanResult r in results) {
                  devices.add(r.device);
                  debugPrint('${r.device.name} found! rssi: ${r.rssi}');
                }
                // LOOK FOR THE SCANNED DEVICES AND PAIR IT
                port = devices.firstWhereOrNull((d) =>
                    d.name == portConfig['portName']); // 'BT578_BLE_DED6');
                if (port != null) {
                  // OPEN PORT AND CONNECT
                  await port!.connect().then((value) async {
                    // CHECK FOR SERVICES
                    services = await port!.discoverServices();
                    // INITIALIZE SERVICE FOR GUID BT578_BLE_DED6
                    service = services.firstWhereOrNull((s) =>
                        s.uuid == Guid('0000ffe0-0000-1000-8000-00805f9b34fb'));
                    // INITIALIZE CHARACTERISTICS FOR GUID BT578_BLE_DED6
                    c = service!.characteristics.firstWhereOrNull((c) =>
                        c.uuid == Guid('0000ffe1-0000-1000-8000-00805f9b34fb'));
                    if (c != null) {
                      // SET NOTIFICATIONS TO ON
                      await c!.setNotifyValue(true);

chipweinberger, 2023-05-19:
please open a PR if you fix the problem

chipweinberger, 2023-05-19:
also, try updating to flutter blue plus  1.5.0

dulifu, 2023-05-19:
I can't update flutter_blue_plus for now, we can't find flutter_blue_plus 1.5.0 using pub get, the error message is as follows: 
```
Because common_packages depends on flutter_blue_plus ^1.5.0 which doesn't match any versions, version solving failed.
pub get failed
```

@chipweinberger

chipweinberger, 2023-05-23:
shows as published for me: https://pub.dev/packages/flutter_blue_plus

chipweinberger, 2023-07-20:
please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger