chipweinberger/flutter_blue_plus #293: CoreBluetooth API MISUSE

codepgq, 2023-05-19:
`[CoreBluetooth] API MISUSE: <CBCentralManager: 0x281fc8000> can only accept this command while in the powered on state
`
Every time I open the app, I will be prompted with the above information

SDK vresion: flutter_blue_plus: ^1.5.0
Dart Version: 3.10.1

chipweinberger, 2023-05-20:
thanks for filing. feel free to open a PR if you fix it.

chipweinberger, 2023-05-20:
I have not been able to repro this, so you will need to debug it yourself.

codepgq, 2023-05-20:
What I found here is different from your sample code is that I didn’t check whether the Bluetooth is turned on, and started searching for devices directly. When I also checked the Bluetooth status, this problem would not occur.

chipweinberger, 2023-07-20:
great that you solved it.

closed 2023-07-20 by chipweinberger