chipweinberger/flutter_blue_plus #294: Scan future hangs indefinitely after 1.5.0 update

chasersmith, 2023-05-19:
After updating to the latest 1.5.0 version, the flutter_blue_plus startScan() Future function never completes/finishes even after stop_scan being logged in the console. For whatever reason, when debugging line by line through the flutter_blue_plus code, it seems to work fine and complete the future after it finishes scanning.

Didn't change anything in my scanning code. Am I missing something that was changed in 1.5.0 that I need to make changes for?

My code
```
Future<List<BluetoothDevice>> scan(Duration duration) async {
    List<BluetoothDevice> scannedDevices = [];

    // Listen to scan stream and add each discovered device to list
    _flutterBluePlus.scanResults.listen((List<ScanResult> results) {
      for (final result in results) {
        final device = result.device;
        if (!scannedDevices.contains(device) && device.name.isNotEmpty) {
          scannedDevices.add(device);
        }
      }
    });

    // Scan for [duration]
    await _flutterBluePlus.startScan(timeout: duration); // <-- THIS IS HANGING

    return scannedDevices;
 }
```

Using Flutter 3.10.1 and Dart 3.0.1

chipweinberger, 2023-05-19:
thanks for reporting.

I have not experienced this issue.

Can you try adding logs to the flutter blue plus code?

(you may need to copy the entire repository to your local disk and then add the dependency by filepath in pubspec.yaml)

chipweinberger, 2023-05-19:
there are likely some bugs in 1.5.0 because it was a rather large change.

chasersmith, 2023-05-19:
The logs seem standard, they signify the scan is stopping just not programmatically ending the Future or something. The "parseFromBytes" and "first manudata for manu ID" lines repeat for as long as the scan goes, I just excluded the duplicate lines.
```
D/BluetoothAdapter(14650): STATE_ON
D/BluetoothLeScanner(14650): Start Scan
D/BluetoothAdapter(14650): STATE_ON
D/BluetoothLeScanner(14650): onClientRegistered() - status=0 clientIf=6 mClientIf=0
D/ScanRecord(14650): parseFromBytes <-- T
D/ScanRecord(14650): first manudata for manu ID
D/BluetoothAdapter(14650): STATE_ON
D/BluetoothLeScanner(14650): Stop Scan
```

chipweinberger, 2023-05-19:
> but it seems like the scan logic is not programmatically ending the Future or something. 

Can you add more logs to debug this specifically? Where is the code getting stuck exactly?

chasersmith, 2023-05-20:
First, my scan function gets called to handle and await the actual scan call for the flutter_blue_plus instance.
[image]

Stepping into this function shows the Future that is being asynchronously waited for within the flutter_blue_plus.dart file inside the package.
[image]

This awaits the scan() function on line 216 where I believe the hanging is happening. Putting a breakpoint on line 224 never gets hit showing that the startScan() Future never completes/returns a value.

The logs I provided in my previous comment are all I get when scanning except, when I debug, I occasionally get this message which I am unsure of being related.
```
D/BluetoothLeScanner(11109): could not find callback wrapper
```

chipweinberger, 2023-05-20:
Where in the `scan` method are we when you are hanging?

I would add logs inside the `scan` method, all over the place.

chipweinberger, 2023-05-20:
I also did push a change to `master`, not to fix this bug, but perhaps it could be related.

chasersmith, 2023-05-20:
A quick note, Inside the ```scan``` method, if I put a break point on line 195 where the scan result stream returns ```result``` and step through each return for the devices it finds, it works properly but if I remove the breakpoint, it no longer works.
[image]

I went ahead and added logs for each of the steps. Here is the code with the log statements to see where they get called.

Starts on line 138
```
Stream<ScanResult> scan({
        ScanMode scanMode = ScanMode.lowLatency,
        List<Guid> withServices = const [],
        List<Guid> withDevices = const [],
        List<String> macAddresses = const [],
        Duration? timeout,
        bool allowDuplicates = false,
    }) async*
    {
        var settings = protos.ScanSettings.create()
        ..androidScanMode = scanMode.value
        ..allowDuplicates = allowDuplicates
        ..macAddresses.addAll(macAddresses)
        ..serviceUuids.addAll(withServices.map((g) => g.toString()).toList());

        if (_isScanning.value == true) {
            throw Exception('Another scan is already in progress.');
        }

        // push to isScanning stream
        _isScanning.add(true);

        if (timeout != null) {
            _scanTimeout = Timer(timeout, () {
                _isScanning.add(false);
                developer.log('stop scan', name: ''); // <-- LOG HERE
                _channel.invokeMethod('stopScan');
            });
        }

        // Clear scan results list
        _scanResults.add(<ScanResult>[]);

        try {
          developer.log('invoke start scan', name: '');// <-- LOG HERE
              await _channel.invokeMethod('startScan', settings.writeToBuffer());
        } catch (e) {
            print('Error starting scan.');
            _isScanning.add(false);
            rethrow;
        }
        developer.log('yield method stream', name: '');// <-- LOG HERE

        yield* FlutterBluePlus.instance._methodStream
            .where((m) => m.method == "ScanResult")
            .map((m) => m.arguments)
            .takeWhile((element) => _isScanning.value)
            .doOnDone(stopScan)
            .map((buffer) => protos.ScanResult.fromBuffer(buffer))
            .map((p) {
                final result = ScanResult.fromProto(p);
                List<ScanResult> list = List<ScanResult>.from(_scanResults.value);
                if (list.contains(result)) {
                    // update advertising data
                    int index = list.indexOf(result);
                    list[index] = result;
                } else {
                    list.add(result);
                }
                _scanResults.add(list);
                developer.log('return result', name: '');// <-- LOG HERE
                return result;
        });
    }

    /// Starts a scan and returns a future that will complete once the scan has finished.
    ///
    /// Once a scan is started, call [stopScan] to stop the scan and complete the returned future.
    ///
    /// timeout automatically stops the scan after a specified [Duration].
    ///
    /// To observe the results while the scan is in progress, listen to the [scanResults] stream,
    /// or call [scan] instead.
    Future startScan({
        ScanMode scanMode = ScanMode.lowLatency,
        List<Guid> withServices = const [],
        List<Guid> withDevices = const [],
        List<String> macAddresses = const [],
        Duration? timeout,
        bool allowDuplicates = false,
    }) async 
    {
        await scan(
            scanMode: scanMode,
            withServices: withServices,
            withDevices: withDevices,
            macAddresses: macAddresses,
            timeout: timeout,
            allowDuplicates: allowDuplicates)
            .drain();
        developer.log('scan finished', name: '');// <-- LOG HERE
        return _scanResults.value;
    }

    /// Stops a scan for Bluetooth Low Energy devices
    Future stopScan() async
    {
        await _channel.invokeMethod('stopScan');
        _scanTimeout?.cancel();
        _isScanning.add(false);
        developer.log('scan stopped', name: '');// <-- LOG HERE
    }
```
The logs were the following (I did remove a bunch of the repeated ```parseFromBytes``` and ```first manudata for manu ID``` messages)

```
[log] invoke start scan
D/BluetoothAdapter(27239): STATE_ON
D/BluetoothLeScanner(27239): Start Scan
D/BluetoothAdapter(27239): STATE_ON
D/BluetoothLeScanner(27239): onClientRegistered() - status=0 clientIf=6 mClientIf=0
[log] yield method stream
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
[log] return result
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
[log] return result
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
[log] return result
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
[log] return result
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
[log] return result
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
D/ScanRecord(27239): parseFromBytes
[log] return result
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
[log] return result
D/ScanRecord(27239): parseFromBytes
D/ScanRecord(27239): first manudata for manu ID
[log] stop scan
D/BluetoothAdapter(27239): STATE_ON
D/BluetoothLeScanner(27239): Stop Scan
```

From this log, it appears that the ``` yield* FlutterBluePlus.instance._methodStream``` stream is returning values but never finishes because the ```stopScan()``` function was never called like it was supposed to as shown by the ``` .doOnDone(stopScan)``` statement attached to the methodStream yield.

chipweinberger, 2023-05-20:
Oh, I did not realize you were using a timeout to end the stream. 

The original implementation used this to end the stream:
`.takeUntil(Rx.merge(killStreams))` (kill streams = listens for the timeout or stopscan)

I removed the RxDart dependency and replaced it with this:
`.takeWhile((element) => _isScanning.value)`

The idea is, when the timeout fires, the stream is supposed to end.

Perhaps you can add a log and dig into why this is not ending the stream.

.takeWhile((element) {
     print("_isScanning: ${_isScanning.value}")
     _isScanning.value
}

chasersmith, 2023-05-20:
That makes sense. For whatever reason it seems the takeWhile is not detecting whenever _isScanning is false. It seems like some kind of synchronization issue because if I put a breakpoint inside the timer when it stops the scan, it works. Not sure what the solution is for this at the moment other than changing how the stream gets stopped. With that being said, I just did my own timeout using Future.delayed inside my function then manually stopped the scan which effectively does the same thing. This is a fine solution in my case, so I don't know how high priority this issue is and if anyone else has experienced it. I appreciate your help and may continue to debug and see if I can find a solution or provide more information.

chipweinberger, 2023-05-20:
I've pushed a change that should fix it: https://github.com/boskokg/flutter_blue_plus/pull/296/files

We just need to close the stream when this timer files. Pretty sure this should work. Let me know.

chasersmith, 2023-05-20:
Tested it, this appears to have fixed the issue. Thank you again for your help and for fixing it so quickly .

closed 2023-05-20 by chasersmith

akeblom, 2023-05-26:
> I've pushed a change that should fix it: https://github.com/boskokg/flutter_blue_plus/pull/296/files
> 
> We just need to close the stream when this timer fires. Pretty sure this should work. Let me know.

Is this included in 1.5.2?

chipweinberger, 2023-05-26:
yes