chipweinberger/flutter_blue_plus #299: setNotifyValue failed

stonehiy, 2023-05-22:
Failed to display in 1.5. x, but there is no problem in 1.4.0，Please help me, thank you！

chipweinberger, 2023-05-22:
please share more logs, ios / android , what is the actual error, etc

wangdongmeng, 2023-05-22:
Same problem, call setNotifyValue(true) and raise an exception

[image]

chipweinberger, 2023-05-22:
Interesting. Yes that is a new check I added. 

Are you on iOS or android?  Ideally, please add logs to the ios obj-c code and android java code so you can inspect why they are returning failure

wangdongmeng, 2023-05-22:
There is a problem with Android, but there is no problem with ios. I try to add the log

wangdongmeng, 2023-05-22:
Need to update the plugin? thank you

chipweinberger, 2023-05-22:
yes ill push an update shortly

chipweinberger, 2023-05-22:
https://github.com/boskokg/flutter_blue_plus/commit/541fa971ed66c9c81bb272fbcfb4da5ad6e7e332

fixed

wangdongmeng, 2023-05-22:
> 

Thanks sir. Verified. You can turn this off

closed 2023-05-22 by chipweinberger