chipweinberger/flutter_blue_plus #300: scan failed

jin-develop, 2023-05-22:
1.5.1 version scan failed in android (flutter version 3.10.1)
please check below log.

I/flutter (29728): Error starting scan.
I/flutter (29728): error : PlatformException(startScan, Field androidScanMode_ for r1.j0 not found. Known fields are [public int r1.j0.h, public x4.y$d r1.j0.i, public boolean r1.j0.j, public x4.y$d r1.j0.k, public static final r1.j0 r1.j0.l, public static volatile x4.x0 r1.j0.m], java.lang.RuntimeException: Field androidScanMode_ for r1.j0 not found. Known fields are [public int r1.j0.h, public x4.y$d r1.j0.i, public boolean r1.j0.j, public x4.y$d r1.j0.k, public static final r1.j0 r1.j0.l, public static volatile x4.x0 r1.j0.m]
I/flutter (29728): 	at x4.s0.f0(Unknown Source:72)
I/flutter (29728): 	at x4.s0.P(Unknown Source:655)
I/flutter (29728): 	at x4.s0.N(Unknown Source:12)
I/flutter (29728): 	at x4.h0.e(Unknown Source:60)
I/flutter (29728): 	at x4.h0.a(Unknown Source:49)
I/flutter (29728): 	at x4.a1.c(Unknown Source:17)
I/flutter (29728): 	at x4.a1.d(Unknown Source:4)
I/flutter (29728): 	at x4.w$a.y(Unknown Source:9)
I/flutter (29728): 	at x4.w$a.x(Unknown Source:4)
I/flutter (29728): 	at x4.w$a.n(Unknown Source:0)
I/flutter (29728): 	at x4.a$a.m(Unknown Source:2)
I/flutter (29728): 	at r1.j.H(Unknown Source:10)
I/flutter (29728): 	at r1.j.x(Unknown Source:2)
I/flutter (29728): 	at r1.j.a(Unknown Source:0)
I/flutter (29728): 	at r1.d.a(Unknown Source:6)
I/flutter (29728): 	at r1.j.p(Unknown Source:49)
I/flutter (29728): 	at r1.j.y(Unknown Source:16)
I/flutter (29728): 	at r1.j.g(Unknown Source:0)
I/flutter (29728): 	at r1.e.a(Unknown Source:6)
I/flutter (29728): 	at r1.j.p(Unknown Source:49)
I/flutter (29728): 	at r1.j.z(Unknown Source:16)
I/flutter (29728): 	at r1.j.h(Unknown Source:0)
I/flutter (29728): 	at r1.f.a(Unknown Source:6)
I/flutter (29728): 	at r1.j.p(Unknown Source:49)
I/flutter (29728): 	at r1.j.onMethodCall(Unknown Source:2070)
I/flutter (29728): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(Unknown Source:17)
I/flutter (29728): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(Unknown Source:18)
I/flutter (29728): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0(Unknown Source:40)
I/flutter (29728): 	at io.flutter.embedding.engine.dart.DartMessenger.a(Unknown Source:0)
I/flutter (29728): 	at io.flutter.embedding.engine.dart.a.run(Unknown Source:12)
I/flutter (29728): 	at android.os.Handler.handleCallback(Handler.java:942)
I/flutter (29728): 	at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter (29728): 	at android.os.Looper.loopOnce(Looper.java:226)
I/flutter (29728): 	at android.os.Looper.loop(Looper.java:313)
I/flutter (29728): 	at android.app.ActivityThread.main(ActivityThread.java:8757)
I/flutter (29728): 	at java.lang.reflect.Method.invoke(Native Method)
I/flutter (29728): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
I/flutter (29728): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
I/flutter (29728): , null) , #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652)
I/flutter (29728): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310)
I/flutter (29728): <asynchronous suspension>
I/flutter (29728): #2      FlutterBluePlus.scan (package:flutter_blue_plus/src/flutter_blue_plus.dart:184)
I/flutter (29728): <asynchronous suspension>
I/flutter (29728): #3      _ConnectionTarget.close.<anonymous closure> (dart:_http/http_impl.dart:2431)
I/flutter (29728): <asynchronous suspension>
I/flutter (29728):

chipweinberger, 2023-05-22:
Thanks for filing. 

To be honest, I'm not sure how to fix this issue or what is causing it. Please feel free to open a PR if you find the root cause.

jin-develop, 2023-05-23:
> 

I fixed it.
it only occured release mode, so need to add proguard rule below.
-keep public class * extends com.google.protobuf.** { *; }

closed 2023-05-23 by jin-develop

politebarista, 2023-05-24:
Hi!
Faced with the problem that all versions above 1.4.0 do not scan devices. In my case, no error other than "D/BluetoothLeScanner(15571): could not find callback wrapper" is output.
Flutter version 3.3.10.

kalismeras61, 2023-06-12:
> Hi @jin-develop @chipweinberger! Faced with the problem that all versions above 1.4.0 do not scan devices. In my case, no error other than "D/BluetoothLeScanner(15571): could not find callback wrapper" is output. Flutter version 3.3.10.

Did you fixed your issue, i am facing same issue 1.4.0 above.

Device not connecting.