chipweinberger/flutter_blue_plus #301: write data and read callback data ,but error occurred!

Goallying, 2023-05-22:
MyCode :

        await c.write([0xab, 0x00, 0x04, 0x00, 0x02, 0xc1, 0xaf],
                                withoutResponse: false);
         var value = await c.read();
         print("onWritePressed ,value =$value");

Reslut ：
[image]

chipweinberger, 2023-05-22:
Please look at the logs, there should be a more detailed error.

see this line:         NSLog(@"[FBP-iOS] didWriteValueForCharacteristic: [Error] %@", [error localizedDescription]);

```
- (void)peripheral:(CBPeripheral *)peripheral
    didWriteValueForCharacteristic:(CBCharacteristic *)characteristic
                             error:(NSError *)error
{
    if (error) {
        NSLog(@"[FBP-iOS] didWriteValueForCharacteristic: [Error] %@", [error localizedDescription]);
    } else if (_logLevel >= debug) {
        NSLog(@"[FBP-iOS] didWriteValueForCharacteristic");
    }

    ProtosWriteCharacteristicRequest *request = [[ProtosWriteCharacteristicRequest alloc] init];
    [request setRemoteId:[peripheral.identifier UUIDString]];
    [request setCharacteristicUuid:[characteristic.UUID fullUUIDString]];
    [request setServiceUuid:[characteristic.service.UUID fullUUIDString]];

    ProtosWriteCharacteristicResponse *result = [[ProtosWriteCharacteristicResponse alloc] init];
    [result setRequest:request];
    [result setSuccess:(error == nil)];

    [_channel invokeMethod:@"WriteCharacteristicResponse" arguments:[self toFlutterData:result]];
}
```

chipweinberger, 2023-05-22:
or android:       `log(LogLevel.DEBUG, "[onCharacteristicWrite] uuid: " + characteristic.getUuid().toString() + " status: " + status);`

```
    @Override
    public void onCharacteristicWrite(BluetoothGatt gatt, BluetoothGattCharacteristic characteristic, int status) {
      log(LogLevel.DEBUG, "[onCharacteristicWrite] uuid: " + characteristic.getUuid().toString() + " status: " + status);
      Protos.WriteCharacteristicRequest.Builder request = Protos.WriteCharacteristicRequest.newBuilder();
      request.setRemoteId(gatt.getDevice().getAddress());
      request.setCharacteristicUuid(characteristic.getUuid().toString());
      request.setServiceUuid(characteristic.getService().getUuid().toString());
      Protos.WriteCharacteristicResponse.Builder p = Protos.WriteCharacteristicResponse.newBuilder();
      p.setRequest(request);
      p.setSuccess(status == BluetoothGatt.GATT_SUCCESS);
      invokeMethodUIThread("WriteCharacteristicResponse", p.build().toByteArray());
    }
```

Goallying, 2023-05-23:
> Please look at the logs, there should be a more detailed error.
>
> see this line: ` NSLog(@"[FBP-iOS] didWriteValueForCharacteristic: [Error] %@", [error localizedDescription]);`
>
> ```
> - [...]
Thank you, I fix this error by changing "writeWithResponse" to "writewithoutResponse" , but the "didWriteValueForCharacteristic" method not called when I had send some data.

chipweinberger, 2023-05-24:
that error message means you are writing to a characteristic that does not allow writing.

it should mean there is a bug in your program.

chipweinberger, 2023-05-24:
> but the "didWriteValueForCharacteristic" method not called when I had send some data.

you will need to debug that yourself. I can't know the reason.

ZameerHaider, 2023-05-24:
but it writes when I send a single line to the same printer

chipweinberger, 2023-05-24:
> but it writes when I send a single line to the same printer

?

Goallying, 2023-05-25:
> that error message means you are writing to a characteristic that does not allow writing.
> 
> it should mean there is a bug in your program.

closed 2023-05-25 by Goallying

JDongKhan, 2023-07-05:
I also got the same issue for iOS.  

didWriteValueForCharacteristic: [Error] Writing is not permitted.