chipweinberger/flutter_blue_plus #302: Scanning when screen is off?

PowerOfFool, 2023-05-23:
Hello,

I have a question regarding the usage of flutter_blue_plus in an Android environment. I am running a foreground service using flutter_background_service that performs scanning. However, when the screen is turned off, the callback function for listening stops being invoked (I believe the scanning is halted). Here are some additional details:

- Operating System: Android
- The battery optimization is disabled.
- I have enabled the retrieval of location information in the background.
- When the app is in the background with the screen on, the scanning continues. However, as soon as the screen is turned off, the scanning stops.

Could you please guide me on how to keep the scanning active even when the screen is off?

Thank you.

chipweinberger, 2023-05-24:
I don't think this is possible in Flutter Blue Plus. You would need to do a lot more research and submit a PR.

xbug42, 2023-06-01:
I can say it is possible because I'm doing exactly that : using flutter_blue_plus from a foreground service (flutter_background_service). 
@PowerOfFool, I'm in the very same situation as you, I don't know why it does not work for you. 

Note that the service kinda works in background but for a limited varying time (sometimes 10 minutes, sometimes 2 hours). Android is shutting down the service eventually. It does not happen in foreground in my case (you have to provide android notifications in foreground).

PowerOfFool, 2023-06-01:
@xbug42 Glad to find someone in the same situation!
From what I have found, I can certainly say that the service is working as the streaming of scan results stops _only_ while the screen is off and resumes when the screen is turned on.

I want the scan results streaming even while the screen is off, how do you implement this? Could you please show me a sample code?

xbug42, 2023-06-01:
@PowerOfFool, sorry I wrote something wrong : I don't make scans in the service, I'm using a connection already established when the screen was on. 

I just tried quickly and experienced the same behavior as you : as long as the screen is off, everything seems to be fine (I can see the usual flutter_blue_plus logs) unless no results are returned :(

Sorry for the hope :(

chipweinberger, 2023-07-20:
AFAIK, *scanning* in background is not possible in FBP.

Feel free to submit a PR if you add support. Feel free to reopen the issue.

closed 2023-07-20 by chipweinberger