chipweinberger/flutter_blue_plus #306: i'm facing this issue using blue plus

criteriontechFlutter, 2023-05-25:
`FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue_plus:generateDebugProto'.
> java.io.IOException: Cannot run program "C:\Users\izhar\.gradle\caches\modules-2\files-2.1\com.google.protobuf\protoc\3.18.0\61d7f5d8849eb1c9b86ff07ca3f6172ce904ed4\protoc-3.18.0-osx-x86_64.exe": CreateProcess error=216, This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher
`

karthikraja001, 2023-06-07:
Same problem for me also

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue_plus:generateDebugProto'.
> java.io.IOException: Cannot run program "C:\Users\HP\.gradle\caches\modules-2\files-2.1\com.google.protobuf\protoc\3.18.0\61d7f5d8849eb1c9b86ff07ca3f6172ce904ed4\protoc-3.18.0-osx-x86_64.exe": CreateProcess error=216, This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher

It specifically downloads the OSX version of executable file instead of EXE file. Please provide the solution for windows.

MrCsabaToth, 2023-07-18:
Now that ProtoBuf is not used any more (see #361 and #339) are you able to compile?

chipweinberger, 2023-07-20:
protobufs were removed. 

please try 1.7.6+, reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger