chipweinberger/flutter_blue_plus #310: withoutResponse only write 20bits?

guduqiucai, 2023-05-26:
c.writeChar.write([0x12, 0x34, ......]); 
can write than 100 characters, but slow in speed, why neen about 3 seconds?
c.writeChar.write([0x12, 0x34, ......], withoutResponse: true); 
only write 20 characters, but fast in speed, why neen about 3 milliseconds?
I need to unpack and send packages of around 1M, how should I write the sending of each package?

brain9520, 2023-05-30:
The amount of written data depends on whether the BLE device is 5.0 or below. If the BLE device is 5.0, it can write up to 250 bytes at a time. If it is 4.0, it can write up to 20 bytes at a time.

chipweinberger, 2023-07-20:
^ this is correct. You can increase the mtu size to try and get faster data rate.

closed 2023-07-20 by chipweinberger