chipweinberger/flutter_blue_plus #313: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null) on Android Device

saguado69, 2023-05-30:
Hi, 
My app works pretty fine on iOS but on Android, I'm getting this issue whenever I poll an external device that returns me >120 bytes length data packet, the first poll never fails, but on the second and third it always rise a PlatformException write_characteristic_error. This does not happen on iOS. I tried to set the MTU to 512 (device default is 20), but still getting the error. I'm using 1.4.0 package version.

What can be the issue?

Debug Console:
I/flutter (10731): 14:54:47:001 > tx: >IQD0001S00U00000AD9A
D/FlutterBluePlugin(10731): mDevices size: 1
D/FlutterBluePlugin(10731): [onCharacteristicWrite] uuid: 0000ffe1-0000-1000-8000-00805f9b34fb status: 0
D/FlutterBluePlugin(10731): [onCharacteristicChanged] uuid: 0000ffe1-0000-1000-8000-00805f9b34fb
I/flutter (10731): 14:54:47:848 < rx: >ISM592S01E2S02E2S03E2S04E2S05E2S06E2S07E2S08E2S09E2S10E2S11E2S12E2S13E2S14E2S15E2S16E2S17E2S18D2S19E2S20E2M01E2M02E+7E6B
I/flutter (10731): 14:54:48:189 > tx: >IACK
I/flutter (10731): 14:54:48:190 > tx: >IQD0002M02U000005C3F
D/FlutterBluePlugin(10731): mDevices size: 1
I/minim_(10731): Background concurrent copying GC freed 155803(4240KB) AllocSpace objects, 3(80KB) LOS objects, 49% free, 13MB/27MB, paused 314us,131us total 121.607ms
I/flutter (10731): PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
D/FlutterBluePlugin(10731): [onCharacteristicWrite] uuid: 0000ffe1-0000-1000-8000-00805f9b34fb status: 0
I/minim_(10731): Background young concurrent copying GC freed 340499(10047KB) AllocSpace objects, 36(900KB) LOS objects, 27% free, 19MB/27MB, paused 210us,110us total 104.242ms

chipweinberger, 2023-05-31:
1.4 is no longer supported

chipweinberger, 2023-07-20:
there are lots of reasons a characteristicWrite can fail. 

see: https://github.com/boskokg/flutter_blue_plus/issues/385

please try 1.7.5

closed 2023-07-20 by chipweinberger