chipweinberger/flutter_blue_plus #314: connect without scaning first throws error

jessewee, 2023-05-31:
1. 
I have connected a ble device and saved the id before
when I relaunch app and call connect directly, it throws PlatformException(connect, Peripheral not found, null, null)
if I start scan after relaunch, then call connect, it works fine
I added the follow code to FlutterBluePlusPlugin.m, It works too
Is there any hidden trouble about this?

[image]

2.
The data went wrong when write continously with version 1.5.2,  but it is normal with 1.4.0

chipweinberger, 2023-05-31:
please read Apples docs / ask ChatGPT

jessewee, 2023-06-01:
> what does the remoteId look like?
> 
> is it a 128-bit uuid?

[image]
This's the id, it comes from ScanResult.

Here's the situation
Our app scans and connects the device, then we save the device id and other infomation, so we can connect directly next time.
I noticed that the comment of BluetoothDevice.fromId said it works only on Android, but I tried on iOS and it works too, so it meets with our demand, util we came cross the problem up there.
We dont feel very reliable since it is kind of not an officially function supported by you authors, and we are unfamiliar with ios development.

Maybe you can provide an official way about this situation? We think it could be an common demand like this.

chipweinberger, 2023-06-01:
please read the iOS bluetooth documentation about reusing IDs.

if it is supported by ios, open a PR and I will add this feature.

chipweinberger, 2023-07-20:
this is now supported on 1.7.5+. reopen new issue if you have issues.

closed 2023-07-20 by chipweinberger

YudizAndroidHardik, 2023-10-05:
In 1.16.2 version I am still getting an exception

`BluetoothDevice.fromId("remoteId);
`
Is it resolved? How can I use this in iOS?

chipweinberger, 2023-10-05:
1. you just scan first.
2. the save remoteId to a file

now you can connect without scanning using the remoteId.

YudizAndroidHardik, 2023-10-05:
For Example, I already have remoteId as a string in my local storage 

Still scanning is required?

chipweinberger, 2023-10-05:
no scanning should not be required then.

YudizAndroidHardik, 2023-10-06:
@chipweinberger Thank you for your response.

Maybe I missed something, below is my simple function, please let me know If I missed something 

```
void initDevice() {
    String remoteId = "AD8D4E01-ACE1-D13F-8520-039DAC3E5362";
    BluetoothDevice device = BluetoothDevice.fromId(remoteId);
    device.connect();
}
```
And here is the exception I found

```
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(connect, Peripheral not found, AD8D4E01-ACE1-D13F-8520-039DAC3E5362, null)

#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
<asynchronous suspension>

#2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:394:13)
<asynchronous suspension>

#3      BluetoothDevice.connect (package:flutter_blue_plus/src/bluetooth_device.dart:63:30)
<asynchronous suspension>
```

**Package Detail:**
_flutter_blue_plus: ^1.16.2_

**Device Detail:**
_iPhone 12 (17.0.3)_

chipweinberger, 2023-10-06:
in order to connect the device must be discoverable 

is it found when scanning?

to be honest you should just always scan for simplicity.