chipweinberger/flutter_blue_plus #319: Cannot obtain list of connected devices on macOS

jeffreywolberg, 2023-06-04:
I have the following connected devices to my M1 MacBook pro
[image]

However, doing 
```
List<BluetoothDevice> bluetoothDevices = await FlutterBluePlus.instance.connectedDevices;
```
returns an empty list.

This seems like a pretty important feature to have working.

Here's the relevant Flutter output from `flutter doctor`
[✓] Flutter (Channel stable, 3.10.3, on macOS 13.3.1 22E772610a darwin-arm64, locale en-US)

chipweinberger, 2023-06-05:
I think you need to scan and connect with FBP.

I don't see this being implemented.

What is your use case?

chipweinberger, 2023-07-20:
please try 1.7.5+. reopen if you have issues.

macOS code is near identical to IOS now.

closed 2023-07-20 by chipweinberger

chipweinberger, 2023-07-26:
speakers and headphones do not support BLE, only Bluetooth Classic.

They will not be usable with this app.