chipweinberger/flutter_blue_plus #322: The bluetooth state code are not working

devgauge888, 2023-06-06:


kalismeras61, 2023-06-13:
> If you manage to fix the bug, please open a PR
> 
> You can also try reverting to 1.4.0

I have same problem, the package broken after 1.4.0. What's changed?

chipweinberger, 2023-06-13:
I have not reproduced the problem.

Have you tried debugging it?

kalismeras61, 2023-06-13:
> I have not reproduced the problem.
> 
> Have you tried debugging it?

It doesn't give any error, 

I am using xcode 15 with iOS 17 beta, if I use the old version 1.4.0 I get the following problem for xcode, if I use version 1.6.0 it does not work and does not connect to the ble device. 

dyld[1858]: Symbol not found: _OBJC_CLASS_$_FlutterError
  Referenced from: <A04AF7CC-E4BE-31AE-BBCB-F3C481FE0035> /private/var/containers/Bundle/Application/BAB70855-ED86-4139-9106-9D93032E1A63/Runner.app/Runner
  Expected in: <8C640A44-12E1-3D9B-A2AE-32638B4BC302> /private/var/containers/Bundle/Application/BAB70855-ED86-4139-9106-9D93032E1A63/Runner.app/Frameworks/flutter_blue_plus.framework/flutter_blue_plus

chipweinberger, 2023-06-13:
Try adding logging statements all over the relevant code in both the Dart and Obj-C.

Try and figure out what is happening.

SilentlyYc, 2023-06-15:
Device Bluetooth is on, I confirm. Using the example in the project, it shows that Bluetooth is not turned on. I don't know what happened.

gorychama, 2023-06-20:
the state of Bluetooth is always OFF ,  only when you turn on your system the  bluetooth service to off and back to on then it work.
this is only on > 1.4.0 
i ve solved the problem temporarily by downgrading the package to 1.3.1,

chipweinberger, 2023-07-20:
i think this was fixed on 1.7.5+. reopen if you have issues.

closed 2023-07-20 by chipweinberger