chipweinberger/flutter_blue_plus #323: On Android, the flutter_blue_plus stock example crashes with exception when connecting to bluez-peripheral Python stock example

jmark1m, 2023-06-06:
Exception occurs after connection and search for services is complete.

The exception thrown from the flutter_blue_plus example is,

`D/BluetoothGatt( 9357): onServiceChanged() - Device=74:04:F1:A3:53:F8
D/BluetoothGatt( 9357): discoverServices() - device: 74:04:F1:A3:53:F8
D/BluetoothGatt( 9357): onConnectionUpdated() - Device=74:04:F1:A3:53:F8 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 9357): onSearchComplete() = Device=74:04:F1:A3:53:F8 Status=0
D/FlutterBluePlugin( 9357): [onServicesDiscovered] count: 4 status: 0
D/BluetoothGatt( 9357): readRssi() - device: 74:04:F1:A3:53:F8
E/JavaBinder( 9357): *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
E/JavaBinder( 9357): java.lang.StackOverflowError: stack size 991KB
E/JavaBinder( 9357):    at com.google.protobuf.GeneratedMessageLite.<init>(GeneratedMessageLite.java:60)
E/JavaBinder( 9357):    at com.boskokg.flutter_blue_plus.Protos$CharacteristicProperties.<init>(Protos.java:7404)
E/JavaBinder( 9357):    at com.boskokg.flutter_blue_plus.Protos$CharacteristicProperties.dynamicMethod(Protos.java:8050)
E/JavaBinder( 9357):    at com.google.protobuf.GeneratedMessageLite.dynamicMethod(GeneratedMessageLite.java:256)
E/JavaBinder( 9357):    at com.google.protobuf.GeneratedMessageLite$Builder.<init>(GeneratedMessageLite.java:350)
E/JavaBinder( 9357):    at com.boskokg.flutter_blue_plus.Protos$CharacteristicProperties$Builder.<init>(Protos.java:7757)
E/JavaBinder( 9357):    at com.boskokg.flutter_blue_plus.Protos$CharacteristicProperties$Builder.<init>(Protos.java:7750)
E/JavaBinder( 9357):    at com.boskokg.flutter_blue_plus.Protos$CharacteristicProperties.dynamicMethod(Protos.java:8053)
E/JavaBinder( 9357):    at com.google.protobuf.GeneratedMessageLite.dynamicMethod(GeneratedMessageLite.java:256)
E/JavaBinder( 9357):    at com.google.protobuf.GeneratedMessageLite.createBuilder(GeneratedMessageLite.java:182)
E/JavaBinder( 9357):    at com.boskokg.flutter_blue_plus.Protos$CharacteristicProperties.newBuilder(Protos.java:7741)
E/JavaBinder( 9357):    at com.boskokg.flutter_blue_plus.ProtoMaker.from(ProtoMaker.java:177)
`

The bluez-peripheral python sample code is pretty straightforward,

https://pypi.org/project/bluez-peripheral/

It is run from a laptop running latest LTS Ubuntu.

jmark1m, 2023-06-06:
I am using,
Flutter 3.10.3 • channel stable • https://github.com/flutter/flutter.git
Framework • revision f92f44110e (5 days ago) • 2023-06-01 18:17:33 -0500
Engine • revision 2a3401c9bb
Tools • Dart 3.0.3 • DevTools 2.23.1

and 
Python 3.10.6

Thank.

MrCsabaToth, 2023-07-18:
Now that ProtoBuf is not used any more (see #361 and #339) can you test this use-case again?

chipweinberger, 2023-07-20:
please try 1.7.5+. reopen if you have issues.

bd-corva, 2023-07-31:
I am still having an issue with the sample code from bluez_peripheral. From what i can tell bluez_peripheral adds the service it creates as a included service(with no way to disable), so when we go and get the included services `service.getIncludedServices()` on Android it crashes because there is no check to validate that included service is not the same as the service it is currently adding, destroying the stack. 

`at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmBluetoothService(FlutterBluePlusPlugin.java:1743)
                                                                                                    	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmBluetoothService(FlutterBluePlusPlugin.java:1743)
                                                                                                    	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmBluetoothService(FlutterBluePlusPlugin.java:1743)
                                                                                                    	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmBluetoothService(FlutterBluePlusPlugin.java:1743)
                                                                                                    	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmBluetoothService(FlutterBluePlusPlugin.java:1743)
                                                                                                    	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$3.onServicesDiscovered(FlutterBluePlusPlugin.java:1332)
                                                                                                    	at android.bluetooth.BluetoothGatt$1$5.run(BluetoothGatt.java:365)
                                                                                                    	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:864)
                                                                                                    	at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback(Unknown Source:0)
                                                                                                    	at android.bluetooth.BluetoothGatt$1.onSearchComplete(BluetoothGatt.java:360)
                                                                                                    	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:168)
                                                                                                    	at android.os.Binder.execTransactInternal(Binder.java:1285)
                                                                                                    	at android.os.Binder.execTransact(Binder.java:1244)`
                                                                                                    	
By adding a simple check [here](https://github.com/boskokg/flutter_blue_plus/blob/e53b1e1261064f6fc6063d3ec074a5dcbc740d98/android/src/main/java/com/boskokg/flutter_blue_plus/FlutterBluePlusPlugin.java#L1736) with something like this:                                                                                                  	

```
            if (s.getUuid().equals(service.getUuid())) {
                continue;
            }
```
Everything works

chipweinberger, 2023-07-31:
fixed: c0388b0b150f6c9fc8685040d20e2b926f4693af

will be included in `1.12.9`

chipweinberger, 2023-08-01:
1.12.9 is now released.

closed 2023-08-01 by chipweinberger

bd-corva, 2023-08-02:
@chipweinberger just a quick update to say it worked for me, i will be testing more and provide other feedback, thx you for the help and the quick turnaround!!!