chipweinberger/flutter_blue_plus #326: Device state disconnect does not fire on listening to the stream

BigSchteve, 2023-06-07:
### Issue
When I attempt to disconnect from a Bluetooth device, the device state does not update to provide me with a disconnect status. Instead the last known status is connected. 

### Steps to reproduce: 

- Scan for BLE devices
- Select a device to listen to
- Subscribe to the device state stream
- Connect to a device
- View state change
- Disconnect from device
- View state change not occurring

### Device Information
Package Version -> 1.5.2
Device -> Galaxy Tab A7 Lite, Android v11

### Example Code
Code we have to perform the connection and disconnection
```dart
final connectedDeviceState = BehaviorSubject<BluetoothDeviceState>();
StreamSubscription<BluetoothDeviceState>? deviceStateSubscription;

Future<void> connect(BluetoothDevice device, {Duration? timeout}) async {
    // Preset to disconnected
    final deviceState = connectedDeviceState.value;
    if (deviceState == BluetoothDeviceState.disconnected) {
      await deviceStateSubscription?.cancel();
      deviceStateSubscription = device.state.listen((state) {
        print('STATE: $state');
        connectedDeviceState.add(state);
      });
      await device.connect(timeout: timeout ?? const Duration(seconds: 5));
      return;
    }
    throw Exception('Cannot connect to an already connected device.');
  }

  Future<void> disconnect(BluetoothDevice device) async {
    final deviceState = connectedDeviceState.value;
    if (deviceState == BluetoothDeviceState.connected) {
      await device.disconnect();
    }
    await deviceStateSubscription?.cancel();
  }
```

chansdad, 2023-07-03:
Same issue , new to flutter , can you provide a bit more detailed explanation on how this is being addressed

chipweinberger, 2023-07-20:
this should be fixed in 1.7.5+

closed 2023-07-20 by chipweinberger