chipweinberger/flutter_blue_plus #328: Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)

ShahzadUmarBaig, 2023-06-08:
Here are the Bluetooth Characteristic Details:

BluetoothCharacteristic{uuid: (UUID Ignore this), deviceId: 04:7F:0E:00:70:85, serviceUuid: (UUID Ignore this), secondaryServiceUuid: null, properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: true, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, descriptors: [], value: []}

and when I try to write data on this Characteristic:
```
characteristic.write([01, 32, 38, 36, 32, 36, 31, 66, 34, 39, 30, 37, 39, 35, 65, 66, 34]);
```

I get the following error:

```

E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(set_notification_error, error when writing the descriptor, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:208:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(set_notification_error, error when writing the descriptor, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:208:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:314:13)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:141:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
I/chatty  (19543): uid=10690(io.blocship.blinq) 1.ui identical 4 lines
E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:314:13)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:141:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
```

Here is the complete function:
```
 Future pairWithBoard(fbp.BluetoothCharacteristic writeCharacteristic, fbp.BluetoothCharacteristic readCharacteristic) async {
      readCharacteristic.setNotifyValue(true);
      readCharacteristic.value.listen((event) {
        print("Read Char: $event");
      });
      writeCharacteristic.write([01, 32, 38, 36, 32, 36, 31, 66, 34, 39, 30, 37, 39, 35, 65, 66, 34]);
}
```

chipweinberger, 2023-07-18:
need more details. try 1.7.4.

closed 2023-07-18 by chipweinberger

chipweinberger, 2023-07-18:
Nothing looks amiss in the code. you'll need to investigate further yourself.

hanskokx, 2023-07-18:
> Nothing looks amiss in the code. you'll need to investigate further yourself.

I'm sure my code isn't quite the same. I'll update this bug when I've got a moment to sit down with it.

hanskokx, 2023-07-18:
Ok, I found/fixed the issue. When trying to `setNotifyValue` on a characteristic, I wasn't `await`ing the result. Simple fix is to `await` the result of `characteristic.setNotifyValue()`.

shwetachauhan-simform, 2023-07-19:
Still facing same issue in version 1.7.5

chan150, 2023-07-20:
Same issue with 1.7.8

chipweinberger, 2023-07-20:
1.7.8 should log more details. what does it log?

chan150, 2023-07-21:
Error messages with 1.7.8:
```
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
<asynchronous suspension>
#2      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:163:9)
<asynchronous suspension>
#3      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:311:7)
<asynchronous suspension>
#4      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:138:5)
<asynchronous suspension>
#5      BLEEcg._sendCommandHolterTime (package:healthrian_ble_support_for_wearecg12/src/mixin/ecg.dart:136:5)
<asynchronous suspension>
#6      BLEEcg.startMeasurementEcg (package:healthrian_ble_support_for_wearecg12/src/mixin/ecg.dart:90:5)
<asynchronous suspension>
#7      _BlePageState.build.<anonymous closure> (package:example/ble/view/ble_page.dart:71:17)
<asynchronous suspension>

      #0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:177:7)
      #1      _rootRunBinary (dart:async/zone.dart:1423:47)
      #2      _CustomZone.runBinary (dart:async/zone.dart:1315:19)
      #3      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
      #4      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
      #5      Future._propagateToListeners (dart:async/future_impl.dart:817:13)
      #6      Future._completeError (dart:async/future_impl.dart:592:5)
      #7      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)
      #8      _rootRun (dart:async/zone.dart:1399:13)
      #9      _CustomZone.run (dart:async/zone.dart:1301:19)
      #10     _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
      #11     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
      #12     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
      #13     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
I/ViewRootImpl(22710): updatePointerIcon pointerType = 1000, calling pid = 22710
D/InputManager(22710): setPointerIconType iconId = 1000, callingPid = 22710
```

Flutter doctor -v:
```
flutter doctor -v
[√] Flutter (Channel stable, 3.10.6, on Microsoft Windows [Version 10.0.19043.1620], locale ko-KR)
    • Flutter version 3.10.6 on channel stable at D:\dev\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision f468f3366c (8 days ago), 2023-07-12 15:19:05 -0700
    • Engine revision cdbeda788a
    • Dart version 3.0.6
    • DevTools version 2.23.1

[√] Windows Version (Installed version of Windows is version 10 or higher)

[√] Android toolchain - develop for Android devices (Android SDK version 33.0.2)
    • Android SDK at C:\Users\ccc\AppData\Local\Android\sdk
    • Platform android-33, build-tools 33.0.2
    • Java binary at: D:\Android\Android Studio\jbr\bin\java
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-9586694)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop for Windows (Visual Studio Build Tools 2019 16.11.26)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools
    • Visual Studio Build Tools 2019 version 16.11.33529.622
    • Windows 10 SDK version 10.0.22621.0

[√] Android Studio (version 2022.2)
    • Android Studio at D:\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-9586694)

[!] Android Studio (version 2023.1)
    • Android Studio at D:\dev\android-studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    X Unable to find bundled Java version.
    • Try updating or re-installing Android Studio.

[√] Connected device (4 available)
    • SM T976N (mobile) • R54NB006ANK • android-arm64  • Android 12 (API 32)
    • Windows (desktop) • windows     • windows-x64    • Microsoft Windows [Version 10.0.19043.1620]
    • Chrome (web)      • chrome      • web-javascript • Google Chrome 114.0.5735.199
    • Edge (web)        • edge        • web-javascript • Microsoft Edge 114.0.1823.82

[√] Network resources
    • All expected network resources are available.

```

chipweinberger, 2023-07-21:
@chan150 I don't see the error message in those logs.

Try 1.8.7 instead.

chan150, 2023-07-21:
> @chan150 I don't see the main error message in those logs, so it's not very useful.
> 
> Try 1.8.7 instead.

It is solved in 1.8.7.
Thank very much @chipweinberger .